espressif/esp-idf #14485: RF coexistence prevents automatic light sleep mode from working

chenlijun99, 2024-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

After I call the appropriate functions to disable wifi and BLE (as described in https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html#wi-fi-bluetooth-and-sleep-modes) and block all my tasks, the MCU should enter in light sleep mode.

### What is the actual behavior?

It doesn't enter in light sleep mode.

### Steps to reproduce.

1. Configure esp_pm to enable automatic light sleep
2. Start Wifi and BLE 
3. Stop Wifi and BLE
4. Observe that the firmware doens't enter in sleep moe

### Debug Logs.

```plain
Actual relevant part of the logs. In short, when coexistence is enabled, even if I stop wifi and BLE, there is still a `esp_timer` from the coexistence module that is running, preventing automatic light sleep mode from working.
The log has been printed using the function `esp_timer_dump`.

I've used `gdb` to watch the address `0x3fcc1c64` and confirmed that it is a timer managed by the `esp_timer` module.

Timer stats:
Name                  Period      Alarm
timer@0x3fcc1c64      0           12895422
```
```

esp-lis, 2024-09-12:
@chenlijun99 perhaps you can use [esp_pm_dump_locks](https://github.com/espressif/esp-idf/blob/master/components/esp_pm/pm_locks.c#L150) to dump the pm lock runtime info. which will help you see the lock holder information.

chenlijun99, 2024-09-12:
I already did that and I didn't see any active pm lock that forbids entering into light sleep mode.

esp-lis, 2024-09-12:
> I already did that and I didn't see any active pm lock that forbids entering into light sleep mode.

can you provide your example code for me?

rac146, 2024-10-15:
I found the same issue while trying to enter automatic light sleep with Wifi and BLE with an ESP32C6.. here are some more scenarios:

1. Enable WIFI Only
2. Disable WIFI
3. Automatic light sleep enters successfully

1. Enable BLE Only and start advertising
2. Disable BLE advertising
3. Automatic light sleep enters successfully

1. Enable BLE and Wifi. Start advertising on BLE
2. Disable BLE advertising
3. Disable Wifi
4. Automatic sleep mode is not entered. The timer referenced from above is still running..

I'm not getting great information using esp_pm_dump_locks - the fact that I'm shutting down the serial port to goto sleep is interfering with getting the info you need. Can someone point me in a direction to better debug this? Definitely an issue with the `esp_coex` module..

rac146, 2024-10-15:
I'm not sure if this helps, but I was able to call the esp_timer_dump with profiling enabled and can see this.. fairly certain this is the timer that is running:

[image]

esp-lis, 2025-03-11:
> I found the same issue while trying to enter automatic light sleep with Wifi and BLE with an ESP32C6.. here are some more scenarios:
>
>
>
>
> I'm not getting great information using esp_pm_dump_locks - [...]

After adding auto light sleep support to the local environment $IDF_PATH/example/bluetooth/nimble/bleprph_wifi_coex/ example, I attempted to reproduce the issue in the following scenario on the ESP32-C6 chip. In my test, automatic sleep worked correctly. Below are my test current waveform diagram and the attachment is my code diff file.

> 1. Enable BLE and Wifi. Start advertising on BLE
> 2. Disable BLE advertising
> 3. Disable Wifi
> 4. Automatic sleep mode is not entered. The timer referenced from above is still running.

[image]

[0001-change-example-auto-light-sleep-support-for-bleprph-.patch](https://github.com/user-attachments/files/19176762/0001-change-example-auto-light-sleep-support-for-bleprph-.patch)

esp-lis, 2025-03-28:
@rac146 Any update for this? I'll close this task for now. If there are any issues, you can reopen it.

closed 2025-03-28 by Alvin1Zhang