espressif/esp-idf #14486: Random mbdetls error -0x2700 (FFFFD900): X509 - Certificate verification failed

QuentinFarizon, 2024-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-Pico-v4

### Power Supply used.

USB

### What is the expected behavior?

I expect to not get mbdetls errors

### What is the actual behavior?

I started gettting mbedtls errors after updating a server certificate.
This code is working on ESP32-S3 without issue, and it was working in ESP32 without issue with a previous server certificate.

I have a task that make call to an https server at interval
It always succeeds to make the call the first time, but then it fails repeatedly, with like 5% of successful calls

On success:
```
I (15:56:02.268) HTTP_CLIENT: URL : https://<username>:@preprod-bokslink-api.boks.app/api/pcbs/4c75258a5a24/available-update
I (15:56:03.300) HTTP_CLIENT: HTTP Status = 200, content_length = 2
```

On error:
```
I (14:32:07.967) HTTP_CLIENT: URL : https://<username>/:@preprod-bokslink-api.boks.app/api/pcbs/4c7525894f90/available-update
0;31mE (14:32:08.108) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (14:32:08.109) esp-tls-mbedtls: Failed to verify peer certificate!
0;31mE (14:32:08.112) esp-tls: Failed to open new connection
0;31mE (14:32:08.118) transport_base: Failed to open a new connection
0;31mE (14:32:08.128) HTTP_CLIENT: Connection failed, sock < 0
0;31mE (14:32:08.131) HTTP_CLIENT: Error perform http request ESP_ERR_HTTP_CONNECT
``` 

Find below the full verbose log

### Steps to reproduce.

In main.c :
`xTaskCreate(&check_and_perform_dfu_task, "check_and_perform_dfu_task", 8192, NULL, 2, NULL);`

[... 523 lines cut ...]

I (21:19:19.497) mbedtls: ssl_msg.c:2370 => flush output

I (21:19:19.503) mbedtls: ssl_msg.c:2384 message length: 7, out_left: 7

I (21:19:19.516) mbedtls: ssl_msg.c:2391 ssl->f_send() returned 7 (-0xfffffff9)

I (21:19:19.518) mbedtls: ssl_msg.c:2418 <= flush output

I (21:19:19.525) mbedtls: ssl_msg.c:3097 <= write record

I (21:19:19.531) mbedtls: ssl_msg.c:5127 <= send alert message

D (21:19:19.537) mbedtls: ssl_tls.c:7446 ! Certificate verification flags 00000008

I (21:19:19.546) mbedtls: ssl_tls.c:3942 <= handshake

E (21:19:19.551) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x2700
I (21:19:19.559) esp-tls-mbedtls: (FFFFD900): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
I (21:19:19.570) esp-tls-mbedtls: Failed to verify peer certificate!
E (21:19:19.578) esp-tls: Failed to open new connection
E (21:19:19.583) transport_base: Failed to open a new connection
E (21:19:19.593) HTTP_CLIENT: Connection failed, sock < 0
E (21:19:19.596) HTTP_CLIENT: Error perform http request ESP_ERR_HTTP_CONNECT
```

QuentinFarizon, 2024-09-05:
Interesting : it happens only if I successfully connect to a MQTT server first, with TLS.

If I don't connect to MQTT, the HTTP SSL connection works perfectly

chegewara, 2024-09-20:
Did you try to check free memory in both cases?
TLS connections require a lot of free memory, close to 40kB IIRC.

AdityaHPatwardhan, 2024-09-30:
Hi @QuentinFarizon 
Can you please provide certificate which started causing this issue?
And yes, Please check the memory available with help of [esp_get_free_heap_size](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/merge_requests/33926/diffs) [esp_get_miniumum_free_heap_size()](https://github.com/espressif/esp-idf/blob/46acfdce969f03c02b001fe4d24fa9e98f6adc5e/components/esp_system/include/esp_system.h#L112) and as @chegewara has mentioned.
If possible, please provide a small reproducible app that can reproduce this issue.
Thanks

AdityaHPatwardhan, 2024-10-08:
@QuentinFarizon Please feel free to re-open if the issue still persists.
Thanks

closed 2024-10-08 by AdityaHPatwardhan