espressif/esp-idf #14487: mqtt5_event_handler changes the contents of event_handler_arg

nopnop2002, 2024-09-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

Chip is ESP32D0WDQ6 (revision 1)

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Development Board

### Power Supply used.

USB

### What is the expected behavior?

mqtt5_event_handler not changes the contents of event_handler_arg

### What is the actual behavior?

mqtt5_event_handler changes the contents of event_handler_arg

### Steps to reproduce.

Make some changes to this sample.

https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt5

1.Registers MQTT event using event_handler_arg
```
    /* The last argument may be used to pass data to the event handler, in this example mqtt_event_handler */
    TaskHandle_t taskHandle[1];
    taskHandle[0] = xTaskGetCurrentTaskHandle();
    ESP_LOGI(TAG, "taskHandle=%"PRIx32, (uint32_t)taskHandle[0]);
    ESP_LOGI(TAG, "taskHandle=%p", &taskHandle);
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt5_event_handler, &taskHandle);
    esp_mqtt_client_start(client);
```

2.Add mqtt5_event_handler
```
static void mqtt5_event_handler(void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
{
    ESP_LOGD(TAG, "Event dispatched from event loop base=%s, event_id=%" PRIi32, base, event_id);
    esp_mqtt_event_handle_t event = event_data;
    esp_mqtt_client_handle_t client = event->client;
    ESP_LOGI(TAG, "mqtt5_event_handler handler_args address=%p", handler_args);
    TaskHandle_t *taskHandle = handler_args;
    ESP_LOGI(TAG, "mqtt5_event_handler taskHandle value=%"PRIx32, (uint32_t)*taskHandle);
    int msg_id;
    switch ((esp_mqtt_event_id_t)event_id) {
    case MQTT_EVENT_CONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_CONNECTED");
        break;
    case MQTT_EVENT_DISCONNECTED:
        ESP_LOGI(TAG, "MQTT_EVENT_DISCONNECTED");
        break;
    case MQTT_EVENT_SUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_SUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_UNSUBSCRIBED:
        ESP_LOGI(TAG, "MQTT_EVENT_UNSUBSCRIBED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_PUBLISHED:
        ESP_LOGI(TAG, "MQTT_EVENT_PUBLISHED, msg_id=%d", event->msg_id);
        break;
    case MQTT_EVENT_DATA:
        ESP_LOGI(TAG, "MQTT_EVENT_DATA");
        break;
    case MQTT_EVENT_ERROR:
        ESP_LOGI(TAG, "MQTT_EVENT_ERROR");
        break;
    default:
        ESP_LOGI(TAG, "Other event id:%d", event->event_id);
        break;
    }
}
```

### Debug Logs.

```plain
I (11490) esp_netif_handlers: example_netif_sta ip: 192.168.10.114, mask: 255.255.255.0, gw: 192.168.10.1
I (11490) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.10.114
I (11500) example_common: Connected to example_netif_sta
I (11500) example_common: - IPv4 address: 192.168.10.114,
I (11510) mqtt5_example: taskHandle address=0x3ffba918  --> this is event_handler_arg address
I (11510) mqtt5_example: taskHandle value=3ffafc50 --> this is event_handler_arg value
I (11520) mqtt5_example: mqtt5_event_handler handler_args address=0x3ffba918 --> address is correct
I (11530) mqtt5_example: mqtt5_event_handler taskHandle value=3ffafc50 ---> value is correct
I (11530) mqtt5_example: Other event id:7
I (12130) mqtt5_example: mqtt5_event_handler handler_args address=0x3ffba918  --> aaddress is correct
I (12130) mqtt5_example: mqtt5_event_handler taskHandle value=3 ---> value is NOT correct
I (12140) mqtt5_example: MQTT_EVENT_CONNECTED
```
```

zawa999, 2024-09-02:
try this
https://bit.ly/4e4zLlq
select the desired compiler in the installer

KaeLL, 2024-09-03:
@zawa999 
> bit.ly

You're joking, right?

snake-4, 2024-09-03:
The event handler isn't modifying the `handler_args`. You're passing it the address of a local variable which likely goes out of scope in the rest of your code.

nopnop2002, 2024-09-03:
>You're passing it the address of a local variable which likely goes out of scope in the rest of your code.

Yes, we are passing the address of the local variable, but we are preventing the local variable from being released as shown below.

```
void app_main(void)
{

    ESP_LOGI(TAG, "[APP] Startup..");
    ESP_LOGI(TAG, "[APP] Free memory: %" PRIu32 " bytes", esp_get_free_heap_size());
    ESP_LOGI(TAG, "[APP] IDF version: %s", esp_get_idf_version());

    esp_log_level_set("*", ESP_LOG_INFO);
    esp_log_level_set("mqtt_client", ESP_LOG_VERBOSE);
    esp_log_level_set("mqtt_example", ESP_LOG_VERBOSE);
    esp_log_level_set("transport_base", ESP_LOG_VERBOSE);
    esp_log_level_set("esp-tls", ESP_LOG_VERBOSE);
    esp_log_level_set("transport", ESP_LOG_VERBOSE);
    esp_log_level_set("outbox", ESP_LOG_VERBOSE);

    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());

    mqtt5_app_start();
    while(1) {
        vTaskDelay(100);
    }
}
```

nopnop2002, 2024-09-03:
>You're passing it the address of a local variable which likely goes out of scope in the rest of your code.

Yes, I'm using a local address, but I'm trying to prevent the local address from being released.

```
    /* The last argument may be used to pass data to the event handler, in this example mqtt_event_handler */
    TaskHandle_t taskHandle[1];
    taskHandle[0] = xTaskGetCurrentTaskHandle();
    //esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt5_event_handler, NULL);
    ESP_LOGI(TAG, "taskHandle address=%p", &taskHandle);
    ESP_LOGI(TAG, "taskHandle value=%"PRIx32, (uint32_t)taskHandle[0]);
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt5_event_handler, &taskHandle);
    esp_mqtt_client_start(client);
    while(1) {
        vTaskDelay(100);
    }
```

nopnop2002, 2024-09-03:
>You're passing it the address of a local variable which likely goes out of scope in the rest of your code.

Yes, you are correct.
The problem was resolved by changing the local address so that it would not be released as shown below.
```
    /* The last argument may be used to pass data to the event handler, in this example mqtt_event_handler */
    TaskHandle_t taskHandle[1];
    taskHandle[0] = xTaskGetCurrentTaskHandle();
    //esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt5_event_handler, NULL);
    ESP_LOGI(TAG, "taskHandle address=%p", &taskHandle);
    ESP_LOGI(TAG, "taskHandle value=%"PRIx32, (uint32_t)taskHandle[0]);
    esp_mqtt_client_register_event(client, ESP_EVENT_ANY_ID, mqtt5_event_handler, &taskHandle);
    esp_mqtt_client_start(client);
    while(1) {
        vTaskDelay(100);
        //ESP_LOGI(TAG, "taskHandle value=%"PRIx32, (uint32_t)taskHandle[0]);
    }
```

closed 2024-09-03 by nopnop2002