espressif/esp-idf #14490: Wifi Connect does not work after flashing => but after manual reset and then no debugging

hf81-ble, 2024-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-S3 WROOM

### Power Supply used.

USB

### What is the expected behavior?

After Flashing and starting debugging the ESP32 should connect to the AP as Station. 

### What is the actual behavior?

the init-> auth process will fail until a RESET will occur. 

After pressing reset the connect and wifi driver will work. But then debugging over native USB is not possible anymore.

### Steps to reproduce.

1. use idf/examples/wifi/softap_sta
2. build
3. debug and run

### Debug Logs.

```plain
Log after Flash and Debug Run

I (33) boot: ESP-IDF v5.3 2nd stage bootloader
I (33) boot: compile time Sep  3 2024 11:27:20
I (33) boot: Multicore bootloader
I (36) boot: chip revision: v0.2
I (40) boot.esp32s3: Boot SPI Speed : 80MHz
I (45) boot.esp32s3: SPI Mode       : DIO
I (49) boot.esp32s3: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=2061ch (132636) map
[... 174 lines cut ...]
I (673) wifi:enable tsf
I (673) wifi:Total power save buffer number: 16
I (673) wifi:Init max length of beacon: 752/752
I (683) wifi:Init max length of beacon: 752/752
I (683) WiFi Sta: Station started
I (693) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (3513) wifi:ap channel adjust o:1,1 n:6,2
I (3513) wifi:new:<6,0>, old:<1,1>, ap:<6,2>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (3523) wifi:state: init -> auth (0xb0)
I (3533) wifi:state: auth -> assoc (0x0)
I (3543) wifi:state: assoc -> run (0x10)
I (3563) wifi:connected with TEST_SSID, aid = 1, channel 6, BW20, bssid = 66:d9:e7:65:ad:ff
I (3563) wifi:security: WPA2-PSK, phy: bgn, rssi: -68
I (3573) wifi:pm start, type: 1

I (3573) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (3573) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 25000, mt_pti: 0, mt_time: 10000
I (3583) wifi:<ba-add>idx:0 (ifx:0, 66:d9:e7:65:ad:ff), tid:0, ssn:0, winSize:64
I (3653) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (4583) WiFi Sta: Got IP:192.168.100.135
I (4583) esp_netif_handlers: sta ip: 192.168.100.135, mask: 255.255.255.0, gw: 192.168.100.254
I (4583) WiFi Sta: connected to ap SSID:TEST_SSID password:12345678
W (4593) esp_netif_lwip: napt disabled on esp_netif:0x3fcb1920
I (4593) main_task: Returned from app_main()
```

hansw123, 2024-09-09:
@hf81-ble 
sorry for late reply,
I can not reproduce your issue in my env
can you reproduce your issue ,if can  could you get another s3 and test it?
from the log，we can not get more information

hansw123, 2024-09-11:
@hf81-ble 
can you provid more info about your device，Did you buy your s3 from official?

closed 2024-10-18 by Sherry616