espressif/esp-idf #14495: Near simultaneous calls to `httpd_ws_send_frame_async` causes frames to become mangled

kevinhikaruevans, 2024-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

NA

### Power Supply used.

USB

### What is the expected behavior?

Calling `httpd_ws_send_frame_async` in rapid succession should correctly queue the packets

### What is the actual behavior?

The frames become mangled and the client(s) force a disconnect. Usually the client shows either a header error, ("Invalid frame header"), fragmentation error (final is incorrectly set to false), or a UTF8 decoding error ("Could not decode a text frame as UTF-8").

### Steps to reproduce.

1. Call `httpd_ws_send_frame_async` to the same fd rapidly
2. Observe that the client (web browser) disconnects due to an error

To resolve the problem, I can wrap the `httpd_ws_send_frame_async` calls in a freertos mutex. However, I think it would make more sense to be handling the mutex in the underlying functions.

### More Information.

I don't have a minimal example right now but I'm noticing this error when broadcasting messages to all ws clients (calling `httpd_get_client_list` and filtering). I'm only sending messages like up to 1-5 messages/sec, but it's enough to cause problems. I'm not sure if it would be better to queue frames or to use a simple mutex.

ESP-YJM, 2024-09-27:
@kevinhikaruevans Could you please follow this example to try?
https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L55

kevinhikaruevans, 2024-10-11:
Hi @ESP-YJM 

I have an example in this branch here: https://github.com/kevinhikaruevans/esp-idf/tree/demo/ws-bug/examples/protocols/http_server/ws_echo_server

Specifically:
- https://github.com/kevinhikaruevans/esp-idf/blob/demo/ws-bug/examples/protocols/http_server/ws_echo_server/src/ws_echo_server.c
- https://github.com/kevinhikaruevans/esp-idf/blob/demo/ws-bug/examples/protocols/http_server/ws_echo_server/src/ws_helper.c

In ws_helper.c, I have a `#define USE_MUTEX`. If this is commented out, it will fail after some time. When it fails, it looks like:

```
$ wscat --connect http://10.0.1.92/ws

< {"index": 2, "time": 4}
< {"index": 0, "time": 4}
< {"index": 1, "time": 4}
< {"index": 2, "time": 4}
< {"index": 0, "time": 4}
< {"index": 1, "time": 4}
< {"index": 2, "time": 4}
error: Invalid WebSocket frame: invalid UTF-8 sequence
```

in Wireshark, this looks like:
[image]

or it might show an errored frame.

When I use the mutex, the program works as expected. But I don't think the application code should be setting the mutex, but the underlying `httpd_ws_send_frame_async` should have a mutex per connection or something similar.

kevinhikaruevans, 2024-10-11:
I think something like this pseudocode would fix the problem, assuming the underlying `send_fn` handles concurrency:

```c
esp_err_t httpd_ws_send_frame_async(...) {
   
   // [...]

   struct sock_db *sess = httpd_sess_get(hd, fd);

   if (!sess) {
      return ESP_ERR_INVALID_ARG;
   }

   size_t frame_length = tx_len + frame->len;
   char frame_buffer = malloc(frame_length);
   
   memcpy(frame_buffer, header_buffer, tx_len);
   memcpy(frame_buffer + tx_len, frame->payload, frame->len);

   if (sess->send_fn(hd, fd, (const char *)frame_buffer, frame_length) < 0) {
      ESP_LOGW(TAG, "Failed to send WS frame");
      return ESP_FAIL;
   }
   
   return ESP_OK;
}
```

ESP-YJM, 2024-10-15:
Yes, this API indeeed exist this issue. But we can avoid directly use it. The API introduction also shows this issue https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L1752. You can use this API `httpd_ws_send_data_async` https://github.com/espressif/esp-idf/blob/master/components/esp_http_server/include/esp_http_server.h#L1804

kevinhikaruevans, 2024-10-17:
I see, should the ws_echo/wss examples be updated to use `httpd_ws_send_data_async` over `httpd_ws_send_frame_async` if the function shouldn't be used?

ESP-YJM, 2024-10-18:
No, as i sent the link previous, the examples use `httpd_ws_send_frame_async`, but it will enqueue it to queue by `httpd_queue_work`. https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/ws_echo_server/main/ws_echo_server.c#L63

So it will not exist the issue.

closed 2024-10-18 by kevinhikaruevans