espressif/esp-idf #14496: The howsmyssl site does not fully support TLS 1.3

gojimmypi, 2024-09-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

While working on adding wolfSSL Certificate Bundle support in https://github.com/wolfSSL/wolfssl/pull/7936 to the Espressif `esp-tls`, I encountered a problem with the [howsmyssl.com](https://www.howsmyssl.com/) web site [used](https://github.com/search?q=repo%3Aespressif%2Fesp-idf%20howsmyssl.com&type=code) in many of the Espressif examples.

When using wolfSSL in a client and forcing only TLS1.3, the connection fails, even though the web site indicates the TLS 1.3 is supported:

[image]

As it turns out, TLS 1.3 is not fully supported. See https://github.com/jmhodges/howsmyssl/issues/716:

> Howsmyssl doesn't yet fully support TLS 1.3. It's working off an old fork of the crypto/tls library from long ago. There's been a tradeoff between upgrading it to a version that supports 1.3 and continuing to have support for the older versions of TLS and ciphersuites that it detects problems in.

I might have said "_Hey, let's use wolfSSL there_", but the site is implemented in the Go Programming Language, and I have little experience with that.

I've initiated some internal conversations at wolfSSL to see how to best proceed. 

Many users choose wolfSSL specifically for the robust TLS1.3 support. This issue is to bring light to a known problem with the sample code in the Espressif ESP-IDF. 

See also https://github.com/espressif/esp-idf/issues/13966

mahavirj, 2024-09-20:
@gojimmypi 

The endpoint that we have in examples is for testing purpose only. It can be overridden as per the end users requirement. If you suggest more suitable endpoint supporting both TLS1.2 and TLS1.3 then we can consider that one as well.

gojimmypi, 2024-09-30:
Hi @mahavirj 

Yes, I understand the `howsmyssl` site is for testing only, but that's also my point: false & misleading test results for TLS 1.3. 

There are alternatives such as https://clienttest.ssllabs.com - a little tricky though, as there's both a port and page redirect.

Other options: www.cloudflare.com or their www.example.com. Also www.letsencrypt.org/docs/staging-environment/  

The `howsmyssl` site is currently hard coded in ESP-IDF example. It is unlikely that end users would consider changing the test site. I also spent some time myself trying to figure out what was wrong. Visiting the web site interactively, it does _appear_ that TLS 1.3 is supported, as noted above.

I don't have a good solution here, particularly for all the already-installed ESP-IDF instances, other than this informative issue.

closed 2025-01-15 by Alvin1Zhang

gojimmypi, 2025-01-15:
It is unfortunate that this was closed as "completed", as the https://github.com/jmhodges/howsmyssl/issues/716 has not been resolved when fully testing TLS connections.

mahavirj, 2025-01-16:
@gojimmypi Just to be clear: what change do you expect on ESP-IDF side here? I understand that the upstream tracker is still open but I don't see any action item on ESP-IDF side. We agreed that there is no other suitable test endpoint which we can consider for ESP-IDF examples.

gojimmypi, 2025-01-17:
@mahavirj it would be good to have a site that properly tests TLS. What requirements are there that are not met by my suggestions, above? At the very least, the documentation and code should indicate the limitations of the current test site.