espressif/esp-idf #14499: Can't use CONFIG_ESP_PHY_ENABLE_CERT_TEST on ESP32-C6

Haerteleric, 2024-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For our inline production tester, we need our DUTs to emit a Continuous wave. For that i want to use the rftest.lib, but it seems incompatible with the ESP32-C6. 

Everytime i add CONFIG_ESP_PHY_ENABLE_CERT_TEST to the sdkconfig file it gets remove on the next build execution.

the lib (rftest.a) and the header for the C6 exist. Why is it not possible to link this lib into our Project ?

mhdong, 2024-09-14:
Hi @Haerteleric 
Currently, our ESP32-C6 does not support cert_test. We have already added cert_test support for ESP32-C6 in the master branch.
commit is dacc1f94dd530eeeafe467b2994085c28a99f3fb, This may take some time to synchronize to GitHub.

mhdong, 2024-09-18:
Hi @Haerteleric 
We have synchronized it to GitHub, please check the commit https://github.com/espressif/esp-idf/commit/aa80dbe2511c235ab2f4632c3e0983f6623ed071. We will synchronize it to other branches as soon as possible.

mhdong, 2024-10-29:
v5.1 https://github.com/espressif/esp-idf/commit/e2822c3a1fdfe9a9b155bcbb178fd4e6402c4e6b
v5.2 https://github.com/espressif/esp-idf/commit/1a1e17a3df205f61baef56bc3eebbe4110e3091c
v5.3 https://github.com/espressif/esp-idf/commit/67b99935d97e0f6c51f6abac7fb267411c95a506

closed 2024-11-13 by Sherry616