espressif/esp-idf #14500: ESP32-S3自己画的PCB板载天线，蓝牙可以正常连接使用，但是WiFi连不上去，这种是天线问题还是软件问题？The ESP32-S3 has its own PCB board antenna, and Bluetooth can be connected and used normally, but WiFi cannot be connected. Is this an antenna issue or a software issue?

caidamao, 2024-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]
我确定WiFi SSID跟密码没有错误，我手头有直接乐鑫淘宝店买的评估开发板。用这个测试是可以连WiFi跟蓝牙的。现在主要是因为蓝牙可用，WiFi不可用，所以很困惑。测试开发板的芯片版本是0.1而自己画的板是0.2版本。我应该去联系商务进行硬件审阅吗？（为了控制变量，我直接使用了IDF自带例程进行的测试。）
I'm sure that the WiFi SSID and password are correct. I have an evaluation board that I bought directly from the Espressif Taobao store. This test can be used to connect WiFi and Bluetooth. Now it is mainly because Bluetooth is available but WiFi is not, so I am very confused. The chip version of the test development board is 0.1, while the board I drew myself is version 0.2. Should I contact the business to conduct a hardware review? To control the variables, I directly used the IDF built-in routine for testing.

caidamao, 2024-09-04:
我用的是IDF5.3
我有测试了蓝牙的连接性能，隔了一堵墙距离在15米以上，电池电量服务每秒上报一次，是正常的。
I'm using IDF 5.3. I have tested the Bluetooth connection performance, and it is normal even with a distance of over 15 meters through a wall, with the battery level service reporting once per second.

caidamao, 2024-09-04:
芯片是立创商城直接机贴的

createskyblue, 2024-09-04:
假如你的程序可以在淘宝上买到的开发板上运行，并且成功连接到WIFI也有蓝牙，但是相同的程序在自己绘制的PCB上无法正常工作（有蓝牙但是没有WIFI），建议可以使用矢量网络分析仪检查一下天线阻抗设计（如NanoVNA）
此外晶振连接到ESP32需要串一个电感，否则可能影响WIFI

caidamao, 2024-09-05:
> 假如你的程序可以在淘宝上买到的开发板上运行，并且成功连接到WIFI也有蓝牙，但是相同的程序在自己绘制的PCB上无法正常工作（有蓝牙但是没有WIFI），建议可以使用矢量网络分析仪检查一下天线阻抗设计（如NanoVNA） 此外晶振连接到ESP32需要串一个电感，否则可能影响WIFI

串了一个电感，然后连成功一次。之后突然又不行了。对脸懵逼

beanjs, 2024-09-05:
我也遇到和你一样的问题了。感觉应该不是天线的问题。我刚才按照他们技术人员的文档抓了包，抓包显示是芯片没有发出Authentication数据包。

MaxwellAlan, 2024-09-26:
Hi @caidamao @beanjs 

As it's a hardware issue rather than something related to ESP-IDF then there is a limit to how much help we can give via this channel. Espressif does offer a hardware design review service
https://www.espressif.com/en/support/services#hardware-rf
, you can contact via the website.

caidamao, 2024-09-26:
> Hi @caidamao @beanjs
>
> As it's a hardware issue rather than something related to ESP-IDF then there is a limit to how much help we can give via this channel. Espressif does offer a hardware design [...]

thanks！

closed 2024-09-26 by caidamao