espressif/esp-idf #14501: How to increase BLE connection Length when using ESP-WrOOM-32

Ajay-rathor, 2024-09-04:
### Is your feature request related to a problem?

I am using ESP-WROOM-32 and ESP-IDF. I have created project using this example:  examples/bluetooth/bluedroid/ble/ble_hid_device_demo

this example is working fine, but it can connect with only one BLE connection at the time.

### Describe the solution you'd like.

I want to increase BLE connection capacity in this example. So it can connect with multiple BLE at same time.

### Describe alternatives you've considered.

I have gone through many ways, checked and modified sdk_config.cmake file , menu-config and the go through program also.

esp-zhp, 2024-09-10:
The ESP32 supports multiple connections, with a maximum of 9 simultaneous connections. To achieve this, both BT_ACL_CONNECTIONS and BTDM_CTRL_BLE_MAX_CONN need to be configured accordingly:

Set BT_ACL_CONNECTIONS to 9
Set BTDM_CTRL_BLE_MAX_CONN to 9

Alvin1Zhang, 2024-10-23:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more details, thanks.