espressif/esp-idf #14504: ESP32S3 SPI transactions assert failed: block_next tlsf_block_functions.h:94 (!block_is_last(block))

lekshith, 2024-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1v

### Espressif SoC revision.

ESP-IDF v5.1.4, chip revision: v0.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

This project Incorporates Ethernet SPI for hosting local web page and SD card for data logging in sdspi mode. Moreover, I have enabled SPIRAM and use it for other dynamic memory allocation. I have 4 RTOS tasks running in the background, 2 for UART operations while other task is the data logging (write to SD) and host a web page for configuration and control. The functionality works well most of the time. Expect it crashes at random times during the logging function dude to shared bus config. The bus is configured to the same host (SPI2_HOST) dude to limited GPIOs.
Initially, the program crashed with looping log meesages as follows:
_```
E (17:36:06.791) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (17:36:06.792) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (17:36:06.795) w5500.phy: w5500_get_link(108): update link duplex speed failed
E (17:36:08.791) w5500.mac: emac_w5500_read_phy_reg(387): read PHY register failed
E (17:36:08.791) w5500.phy: w5500_update_link_duplex_speed(65): read PHYCFG failed
E (17:36:08.795) w5500.phy: w5500_get_link(108): update link duplex speed failed
```_
To which I seeked the solution from https://github.com/espressif/esp-idf/issues/6624 and https://github.com/espressif/esp-idf/issues/8179, which stopped this looping errors.

### What is the actual behavior?

However the patch for the spi_master.c works well for about 45 mins and then started failing with the error as follows:
E (5516841) w5500.mac: emac_w5500_read_phy_reg(335): read PHY register failed
E (5516841) w5500.phy: w5500_update_link_duplex_speed(69): read PHYCFG failed
E (5516841) w5500.phy: w5500_get_link(112): update link duplex speed failed
E (5516901) w5500.mac: w5500_get_rx_received_size(152): read RX RSR failed
E (5518051) w5500.mac: w5500_get_rx_received_size(152): read RX RSR failed

To which I tried a quick fix suggested in https://github.com/espressif/esp-idf/issues/11295 to replace the spi_device_transmit with spi_device_polling_transmit. Which has stopped all the looping error. However the system is now crashing after about hour or so runtime with assert failed: block_next tlsf_block_functions.h:94 (!block_is_last(block)).

### Steps to reproduce.

Here is the complete Ethernet Initialisation
void init_ethernet(void)
{
    // Initialize the SPI mutex
    init_spi_mutex();
    esp_netif_config_t netif_cfg_DHCP_client = ESP_NETIF_DEFAULT_ETH();
    esp_netif_t *eth_netif = esp_netif_new(&netif_cfg_DHCP_client);

    spi_bus_config_t buscfg = {
[... 119 lines cut ...]
0x40378f59: spi_device_polling_transmit at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/driver/spi/gpspi/spi_master.c:1127
0x4202cc95: poll_busy at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/driver/spi/sdspi/sdspi_host.c:576
0x4202d00f: start_command_write_blocks at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/driver/spi/sdspi/sdspi_host.c:927
0x4202d5cd: sdspi_host_start_command at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/driver/spi/sdspi/sdspi_host.c:480
0x4202d922: sdspi_host_do_transaction at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/driver/spi/sdspi/sdspi_transaction.c:150
0x42068553: sdmmc_send_cmd at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/sdmmc/sdmmc_cmd.c:24
0x42017d17: sdmmc_write_sectors_dma at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/sdmmc/sdmmc_cmd.c:448
0x42017ef4: sdmmc_write_sectors at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/sdmmc/sdmmc_cmd.c:412
0x42016477: ff_sdmmc_write at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/diskio/diskio_sdmmc.c:66
0x420162f6: ff_disk_write at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/diskio/diskio.c:98
0x420129ee: sync_window at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/src/ff.c:1065
0x42013445: sync_fs at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/src/ff.c:1117
0x420149cb: f_sync at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/src/ff.c:4199
0x420149f1: f_close at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/src/ff.c:4226
0x42016a1c: vfs_fat_close at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/fatfs/vfs/vfs_fat.c:549
0x42009231: esp_vfs_close at C:/Users/lchandrashekar/esp/v5.1/esp-idf/components/vfs/vfs.c:512 (discriminator 3)
0x4205a68a: close at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32s3-elf/src/newlib/newlib/libc/syscalls/sysclose.c:9
0x420115be: gzclose_w at C:/CompiledProject/SD-V2/main/zlib/gzwrite.c:661
0x42010e50: gzclose at C:/CompiledProject/SD-V2/main/zlib/gzclose.c:21 (discriminator 2)
0x4200d908: log_to_sd_card at C:/CompiledProject/SD-V2/main/sd_card/sd_card.c:788
```

### More Information.

Any suggestions or workarounds are welcome. Thank you.

lekshith, 2024-09-04:
> try this https://gofile.io/d/Lji8KD select the desired compiler in the installer

lol, I'm not running an unknown file just named as fix.exe in the system.
care to explain what that is?

pengkebao, 2025-05-26:
我今天用的idf v5.4和你问题一样，关键一开始是好的，运行大概1个小时左右就出现
E (4151150) w5500.mac: emac_w5500_read_phy_reg(431): read PHY register failed
E (4151150) w5500.phy: w5500_update_link_duplex_speed(69): read PHYCFG failed
E (4151150) w5500.phy: w5500_get_link(112): update link duplex speed failed
E (4153150) w5500.mac: emac_w5500_read_phy_reg(431): read PHY register failed
E (4153150) w5500.phy: w5500_update_link_duplex_speed(69): read PHYCFG failed
E (4153150) w5500.phy: w5500_get_link(112): update link duplex speed failed
E (4155150) w5500.mac: emac_w5500_read_phy_reg(431): read PHY register failed
E (4155150) w5500.phy: w5500_update_link_duplex_speed(69): read PHYCFG failed
E (4155150) w5500.phy: w5500_get_link(112): update link duplex speed failed