espressif/esp-idf #14505: WIFI_SCAN_PARAMS_DEFAULT_CONFIG macro should also set channel bitmap

TD-er, 2024-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

See the define of this macro:
https://github.com/espressif/esp-idf/blob/6673376297b921d438790d195014b860eaf8bb17/components/esp_wifi/include/esp_wifi_types_generic.h#L154C9-L159

This macro does not clear the `channel_bitmap` members, which can cause issues as those will then contain random data.
See related PR in arduino-esp32: https://github.com/espressif/arduino-esp32/pull/10282 and related issue: https://github.com/espressif/arduino-esp32/issues/10281

brentru, 2024-09-12:
@TD-er  I'm seeing/having a similar issue with WiFi Scan as well and this macro is not set properly.

hansw123, 2024-10-17:
@TD-er 
sorry for late reply
WIFI_SCAN_PARAMS_DEFAULT_CONFIG is  used for  esp_wifi_set_scan_parameters() and struct wifi_scan_default_params_t, and only set scan time releated config, bitmap will be set default and will cauese issue

closed 2025-01-15 by Alvin1Zhang