espressif/esp-idf #14508: CONFIG_BOOTLOADER_NUM_PIN_APP_TEST pin configuration for ESP32-S3

Psyjonas, 2024-09-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For device production, we are using the test partition so we can segment the test code from the application code.
To activate the test partition we configure the CONFIG_BOOTLOADER_NUM_PIN_APP_TEST with the same gpio as the UART0 RX pin. This way we can activate the test partition sending a BREAK over UART.

The problem with ESP32-S3 is that the default UART0 RX gpio is the gpio 44 but the Kconfig only supports the range 0-39
This can be easily fixed changing the Kconfig to support custom ranges depending of the target
```
        range 0 39 if IDF_TARGET_ESP32
        range 0 44 if IDF_TARGET_ESP32S3
```

I would like an official fix from espressif.
Can you add a custom range for all the supported targets to the CONFIG_BOOTLOADER_NUM_PIN_APP_TEST and CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET

SoucheSouche, 2024-09-05:
Hello @Psyjonas,
That's a good point. I will update the configs with different ranges for the different targets. Thanks for pointing it out!

SoucheSouche, 2024-09-06:
Hello @Psyjonas,
I updated CONFIG_BOOTLOADER_NUM_PIN_APP_TEST  and CONFIG_BOOTLOADER_NUM_PIN_FACTORY_RESET with target specific ranges. The modifications were merged internally and will be available with the next sync with GH.

closed 2024-09-14 by espressif-bot