espressif/esp-idf #14509: [esp-eye] Station mode wifi disconnected

leehakgun, 2024-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Open the web page in softap mode, store nvs information, and connect the app in station mode with ssid and passwod

### What is the actual behavior?

Open the web page in softap mode, store nvs information, and disconnect the app in station mode with ssid and passwod

### Steps to reproduce.

void Dawon_Init_SoftAP(SysInfo *pSysinfo)
{	

	uint8_t eth_mac[6];
	char mkSsid[32];
	char SetPwd[32];
	
	memset(SetPwd, 0, sizeof(SetPwd));
	
	ESP_ERROR_CHECK(esp_netif_init());
	ESP_ERROR_CHECK(esp_event_loop_create_default());
	esp_netif_create_default_wifi_ap();
	
	wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
	ESP_ERROR_CHECK(esp_wifi_init(&cfg));
	
	setDns();
	ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_APSTA)); //WIFI_MODE_AP
	ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, ESP_EVENT_ANY_ID, &dwd_event_handler, NULL));
	ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, ESP_EVENT_ANY_ID, &dwd_event_handler, NULL));
	
	
	esp_wifi_get_mac(ESP_IF_WIFI_AP, eth_mac);
	if(pSysinfo->DeviceMeasureType == DWD_TYPE_THREE_100A)
	{
		sprintf(mkSsid, DWD_SSID_PREPIX_3P100A"%02x%02x%02x", eth_mac[3], eth_mac[4], eth_mac[5]);
	}else if(pSysinfo->DeviceMeasureType == DWD_TYPE_THREE_400A)
	{
		sprintf(mkSsid, DWD_SSID_PREPIX_3P400A"%02x%02x%02x", eth_mac[3], eth_mac[4], eth_mac[5]);
	}else if(pSysinfo->DeviceMeasureType == DWD_TYPE_THREE_800A)
	{
		sprintf(mkSsid, DWD_SSID_PREPIX_3P800A"%02x%02x%02x", eth_mac[3], eth_mac[4], eth_mac[5]);
	}else
	{
		sprintf(mkSsid, DWD_SSID_PREPIX_1P"%02x%02x%02x", eth_mac[3], eth_mac[4], eth_mac[5]);
	}
	sprintf(pSysinfo->mac_st, "%02x%02x%02x%02x%02x%02x", eth_mac[0], eth_mac[1], eth_mac[2], eth_mac[3], eth_mac[4], eth_mac[5]);
	
[... 568 lines cut ...]
I (964) wifi_init: udp mbox: 6
I (965) wifi_init: tcp mbox: 6
I (969) wifi_init: tcp tx win: 5744
I (973) wifi_init: tcp rx win: 5744
I (977) wifi_init: tcp mss: 1440
I (981) wifi_init: WiFi IRAM OP enabled
I (1027) wifi_init: WiFi RX IRAM OP enabled
ssid: dawonlap
password: iamdawon
I (1028) dwd softAP&Station: Free heap size before Wi-Fi start: 175680 bytes
I (1059) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1155) wifi:mode : sta (24:0a:c4:eb:2c:fc)
I (1156) wifi:enable tsf
I (1158) dwd softAP&Station: Station모드 Wi-Fi 프로토콜 모드가 BGN으로 설정되었습니다.
Received event_base: WIFI_EVENT
wifi 이벤트
I (1166) dwd softAP&Station: [wifi_handler:120] event : WIFI_EVENT_STA_START
ssid:
password:
I (1175) dwd softAP&Station: wifi_init_sta finished.
I (1762) wifi:flush txq
I (1763) wifi:stop sw txq
I (1763) wifi:lmac stop hw txq
Received ets Jun  8 2016 00:22:57
```

hansw123, 2024-09-24:
hi @leehakgun 
sorry for late reply，Can you replace the Korean in the reproduction code with English?
i

hansw123, 2024-09-24:
@leehakgun 
can you describe your issue more detailed？

closed 2025-01-15 by Alvin1Zhang