espressif/esp-idf #14511: i2c sync transmission with i2c_bus_master_xxx functions is slow

eco747, 2024-09-05:
### Is your feature request related to a problem?

The minimum time between 2 sync i2c master transmissions is 10ms (when not changing RTOS time switch).

### Describe the solution you'd like.

sync i2c transmissions are based on events, it should not .
if not using events but direct variable in i2c transaction_done ISR , we can go down to 100us (100x faster) on esp32s3
I know that events are more clean, but the difference is really huge.

### Describe alternatives you've considered.

1. install isr before transmission so i2c transmission become async
2. set a (volatile) flag inside isr
3. call i2c_master_transmit 
4. wait for the flag to be set / or timeout
5. uninstall isr

mythbuster5, 2025-01-20:
Thanks for your advice, but in my test the minimum between 2 sync transaction is not that long. As your suggest, I kept this conservative so far because we will not do big driver refactoring to improve performance because it will cause potential breaking change to other users. But yes, improve performance is what need to be improved, I will keep testing internally.

gstrakabsci, 2026-10-27:
Adding measured data for the multi-task case, ESP32-S3, IDF 5.5.6 (same structure in v6.0), 100 kHz bus, `FREERTOS_HZ=100`, 4 equal-priority tasks hammering one bus (~200k transactions/run):

- **Per-call latency:** worst case 22–40 ms under 4-task contention, up to ~72 ms with concurrent NVS flash activity. The legacy driver on identical traffic: ~9 ms worst case (~30 ms with flash). So the new driver's worst-case call latency is roughly 2–4× the legacy driver's.
- **Uneven bus sharing:** two workers completed ~70k transactions in ~121–128 s while the other two needed ~148 s for ~30k — same priority, same bus. Legacy: all workers finished within ~91 s.
- **Practical impact:** with a 50 ms timeout, calls timed out under load (`ESP_ERR_TIMEOUT`), which is how we found this.

Mechanism, as I read the code: `bus_lock_mux` is a binary semaphore (no priority inheritance) held for the whole transaction, and inside that held lock `s_i2c_send_commands` blocks on `cmd_semphr` once per command plus a final `event_queue` wait — several ISR→task wakeups per transaction all inflating the next waiter's queue time. Is the per-command task wakeup needed, or could the ISR drive the whole job list to completion with a single wake?

gstrakabsci, 2026-11-12:
@mythbuster5 See my recent comment. Are you still investigating this performance issue?