espressif/esp-idf #14514: ESPProvision Wifi Provisioning - Additional BLE Endpoints not working

gadget-man, 2024-09-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1U

### Power Supply used.

USB

### What is the expected behavior?

When adding a custom endpoint using the example code at https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/provisioning/wifi_provisioning.html?highlight=wifi%20provisioning#additional-endpoints, I expect the ESP32-S3 to broadcast that endpoint in addition to the pre-defined prov-* endpoints included as part of the Wifi Provisioning service

### What is the actual behavior?

Everything appears to load OK and no errors are reported, however when inspecting the BLE device using a scanner, the custom endpoint does not appear. Also, calling 'ESPDevice.sendData(...)` from the iOS app returns sendDataError with error `BLE characteristic does not exist`.

### Steps to reproduce.

Here is the code I'm using:

`void provision_ble(void)
{

    if (!wifi_sta_initialised)
    {
        sta_netif = esp_netif_create_default_wifi_sta();
        wifi_sta_initialised = true;
    }
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    /* Configuration for the provisioning manager */
    wifi_prov_mgr_config_t config = {
        .scheme = wifi_prov_scheme_ble,
        .scheme_event_handler = WIFI_PROV_SCHEME_BLE_EVENT_HANDLER_FREE_BTDM};

    /* Initialize provisioning manager with the
     * configuration parameters set above */
    ESP_ERROR_CHECK(wifi_prov_mgr_init(config));

    esp_err_t err = wifi_prov_mgr_endpoint_create("custom-data");
    if (err != ESP_OK)
    {
        ESP_LOGE(TAG, "Error creating custom endpoint: %d", err);
    }

    char *ssid = ipb_config_get_str("wifi.ap.ssid");
    const char *pop = ipb_config_get_str("wifi.ap.pass");
    char service_name[21]; // Ensure the array is large enough to hold the string and null terminator
    strncpy(service_name, ssid, sizeof(service_name) - 1);
    service_name[sizeof(service_name) - 1] = '\0'; // Ensure null termination

    wifi_prov_security_t security = WIFI_PROV_SECURITY_1;
    wifi_prov_security1_params_t *sec_params = pop;

    free(ssid);

    const char *service_key = NULL;

    uint8_t custom_service_uuid[] = {
        /* LSB <---------------------------------------
         * ---------------------------------------> MSB */
        [removed]};

    wifi_prov_scheme_ble_set_service_uuid(custom_service_uuid);

    ESP_ERROR_CHECK(wifi_prov_mgr_start_provisioning(security, (const void *)sec_params, service_name, service_key));

    err = wifi_prov_mgr_endpoint_register("custom-data", prov_sys_reboot, NULL);

    wifi_prov_print_qr(service_name, pop, PROV_TRANSPORT_BLE);
}

esp_err_t prov_sys_reboot(uint32_t session_id, const uint8_t *inbuf, ssize_t inlen,
                          uint8_t **outbuf, ssize_t *outlen, void *priv_data)
{
    ESP_LOGI(TAG, "Calling sys reboot");
    if (inbuf)
    {
        ESP_LOGI(TAG, "Received data: %.*s", inlen, (char *)inbuf);
    }
    char *response = rpc_sys_reboot(NULL);
    *outbuf = (uint8_t *)strdup(response);
    if (*outbuf == NULL)
    {
        ESP_LOGE(TAG, "System out of memory");
        return ESP_ERR_NO_MEM;
    }
    *outlen = strlen(response) + 1; /* +1 for NULL terminating byte */
    free(response);

    return ESP_OK;
}`

### More Information.

Other than the custom endpoint, all other aspects of the BLE provisioning process work correctly.

gadget-man, 2024-09-05:
[image]
[image]

gadget-man, 2024-09-13:
By means of an update, I can get this working on Android, but not on iOS.

My (working) Android call is:
`val byteArray = "forceEmptyBox".toByteArray()
  espDevice?.sendDataToCustomEndPoint("custom-api", byteArray, object : ResponseListener {`

However, when I call the iOS equivalent, I get the error:
`let jsonData = "forceEmptyBox".data(using: .utf8)!
  self.espDevice?.sendData(path: "custom-api", data: jsonData) { responseData, error in`

gadget-man, 2024-09-13:
It turns out that I simply needed to restart the iOS device and scanning then worked correctly.

closed 2024-09-13 by gadget-man

hrushikesh430, 2024-09-14:
> It turns out that I simply needed to restart the iOS device and scanning then worked correctly.

That's great @gadget-man. If any issue comes please feel free to reopen the issue.