espressif/esp-idf #14516: i2c 从站暂停引起cpu无限重启

flair0626, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3.1

### Espressif SoC revision.

ESP32-C2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C2

### Power Supply used.

External 3.3V

### What is the expected behavior?

无限重启

### What is the actual behavior?

看门口超时

### Steps to reproduce.

1. Step 当ESP32作为主站使用i2c_master_transmit_receive 10ms间隔发送数据时
2. Step 将从站mcu停止运行

### Debug Logs.

```plain
E (44955) i2c.master: I2C transaction timeout detected
E (44955) i2c.master: s_i2c_synchronous_transaction(872): I2C transaction failed
E (49955) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (49955) task_wdt: - IDLE (CPU 0)
E (49955) task_wdt: Tasks currently running:
E (49955) task_wdt: CPU 0: emcu
E (49955) task_wdt: Print CPU 0 (current core) backtrace
Print CPU 0 (current core) registers
Core 0 register dump:
MEPC : 0x4202639a RA : 0x42026368 SP : 0x3fcbb850 GP : 0x3fcb12a0
0x4202639a: i2c_ll_master_clr_bus at C:/Users/flair/esp/v5.3.1/esp-idf/components/hal/esp32c2/include/hal/i2c_ll.h:680 (discriminator 1)
(inlined by) s_i2c_master_clear_bus at C:/Users/flair/esp/v5.3.1/esp-idf/components/esp_driver_i2c/i2c_master.c:77 (discriminator 1)
(inlined by) s_i2c_hw_fsm_reset at C:/Users/flair/esp/v5.3.1/esp-idf/components/esp_driver_i2c/i2c_master.c:100 (discriminator 1)
0x42026368: s_i2c_master_clear_bus at C:/Users/flair/esp/v5.3.1/esp-idf/components/esp_driver_i2c/i2c_master.c:77
(inlined by) s_i2c_hw_fsm_reset at C:/Users/flair/esp/v5.3.1/esp-idf/components/esp_driver_i2c/i2c_master.c:100

TP : 0x3fcbb980 T0 : 0x4005e3c8 T1 : 0x00000000 T2 : 0xffffffff
S0/FP : 0x00000007 S1 : 0x3fcbaf54 A0 : 0x3fcbaf64 A1 : 0x3fcbb858
A2 : 0x00000000 A3 : 0x00000800 A4 : 0x60013000 A5 : 0x00000001
A6 : 0x00000080 A7 : 0x42005b9e S2 : 0x3fcbaf64 S3 : 0x00000006
[... 328 lines cut ...]
S8 : 0x00000000 S9 : 0x00000000 S10 : 0x00000000 S11 : 0x00000000
T3 : 0x00000000 T4 : 0x3fcb4102 T5 : 0x00001800 T6 : 0x00000000
MSTATUS : 0x00000000 MTVEC : 0x3fcbaf64 MCAUSE : 0x00000012 MTVAL : 0x0000001f
MHARTID : 0x0000000e
```

### More Information.

在I2C_ll.h中 添加斜体粗体部分代码解决。
static inline void i2c_ll_master_clr_bus(i2c_dev_t *hw, uint32_t slave_pulses)
{
hw->scl_sp_conf.scl_rst_slv_num = slave_pulses;
hw->scl_sp_conf.scl_rst_slv_en = 1;
hw->ctr.conf_upgate = 1;
// hardward will clear scl_rst_slv_en after sending SCL pulses,
// and we should set conf_upgate bit to synchronize register value.
**_int wait_cnt = 0;
while (hw->scl_sp_conf.scl_rst_slv_en)
{
wait_cnt++;
if (wait_cnt > 100000)
break;
};_**
hw->ctr.conf_upgate = 1;
}

mythbuster5, 2024-09-06:
The fix is on the way

mythbuster5, 2024-11-06:
Already fixed.

closed 2024-11-06 by mythbuster5