espressif/esp-idf #14517: esp-idf/examples/peripherals/camera /camera_dsi run failed

hujianfeng, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

master

### Espressif SoC revision.

esp32-p4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-p4-function-ev-board

### Power Supply used.

USB

### What is the expected behavior?

NONE

### What is the actual behavior?

NONE

### Steps to reproduce.

1 build
2 flash

brunoR1000, 2024-09-06:
This might help you
https://bit.ly/3yXXzbX
Passcode: d5kp14h
You must have gcc compiler installed

suda-morris, 2024-09-06:
What do you mean by "run fail"? Have you checked the README file, it should list the LCD and camera device used by the example, maybe they're different from the ones on your board?

hujianfeng, 2024-09-06:
This example failed to run on the ESP32-p4-function-ev-board development board after build and flash it.

suda-morris, 2024-09-06:
Yes, at least, the LCD IC is different from the one on your board. Before we make any updates to the camera example, you can first try the LCD initialization code from [this example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/lcd/mipi_dsi).

Y1hsiaochunnn, 2024-09-10:
Hi @suda-morris ,
I wanna if when the [mipi-csi demo](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/camera/camera_dsi) will support EK79007 display

closed 2024-09-19 by espressif-bot