espressif/esp-idf #14519: `esp-idf/examples/peripherals/dac/dac_continuous/signal_generator` failed in esp32s3-wroom-1

zenny, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should get built, but fails to do so even with `idf.py set-target esp32s3`.

### What is the actual behavior?

Instead it outputs errors as reported in the log posted in https://termbin.com/jpjm

### Steps to reproduce.

1. Copy `esp-idf/examples/peripherals/dac/dac_continuous/signal_generator` to `$HOME/esp` dir.
2. Specify `idf.py set-target esp32s3`
3. Run `idf.py build`
4. You get the error as  in https://termbin.com/jpjm
...

### Build or installation Logs.

```plain
Log posted in https://termbin.com/jpjm
```

suda-morris, 2024-09-06:
Because only ESP32 and ESP32-S2 has the DAC peripheral. On other chips, for a **simple** DAC use case, you can try https://github.com/espressif/esp-idf/tree/master/examples/peripherals/sigma_delta/sdm_dac as an alternative. But the SDM peripheral doesn't support DMA.

zenny, 2024-09-06:
> Because only ESP32 and ESP32-S2 has the DAC peripheral. On other chips, for a **simple** DAC use case, you can try https://github.com/espressif/esp-idf/tree/master/examples/peripherals/sigma_delta/sdm_dac as an alternative. But the SDM peripheral doesn't support DMA.

@suda-morris Thanks for the pointer. I compiled from https://github.com/espressif/esp-idf/tree/master/examples/peripherals/sigma_delta/sdm_dac, but thge output from pinout 0 fluctuates between 10kHz to 28kHz and the duty cycle plummets to 0% once it reaches 20kHz!

Meanwhile, which esp32s3 do you recommend to develop at least a dual channel signal generator hardwarewise? I appreciate if you share any pointer to any library that can be used to create a sweep frequency between 1Hz to 3Mhz which can be changeable remotely either via wifi/bluetooth? Thanks and xiexie.

Cheers,
/zenny

suda-morris, 2024-09-09:
@zenny Have you connected a *low pass filter* to the GPIO output pin?

Since there's no DAC on esp32s3, and I'm afraid the 3MHz is a little bit high for SDM. Maybe you can try some SPI interfaced DAC or DDS module.

suda-morris, 2025-01-17:
> Can esp32s3-wroom can handle say 1.5Mhz? 

I'm afraid not able to...

closed 2025-01-17 by suda-morris