espressif/esp-idf #14520: ./install.sh fails to download the packages due to certificate error

nakajimayoshi, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-2744-g59e1838270

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

installs the packages without error

### What is the actual behavior?

Throws a download error

### Steps to reproduce.

1. Clone the repo 
2. cd esp-idf
3. run ```./install.sh```
...

### Build or installation Logs.

```plain
esp-idf nakajimayoshi$ ./install.sh                                                                                   
Detecting the Python interpreter
Checking "python3" ...
Python 3.10.0
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating /Users/nakajimayoshi/.espressif/idf-env.json
Selected targets are: esp32c61, esp32s3, esp32s2, esp32c3, esp32h2, esp32p4, esp32c5, esp32c2, esp32, esp32c6
Current system platform: macos-arm64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Installing xtensa-esp-elf-gdb@14.2_20240403
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz
Destination: /Users/nakajimayoshi/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
WARNING: Download failure: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz to /Users/nakajimayoshi/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz
Destination: /Users/nakajimayoshi/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
WARNING: Download failure: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz to /Users/nakajimayoshi/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz
Destination: /Users/nakajimayoshi/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
WARNING: Download failure: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:997)>
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz to /Users/nakajimayoshi/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Please make sure you have a working Internet connection.
Certificate issues are usually caused by an outdated certificate database on your computer.
Please check the documentation of your operating system for how to upgrade it.
Running "./Install\ Certificates.command" might be able to fix this issue.
Running "/Library/Frameworks/Python.framework/Versions/3.10/bin/python3 -m pip install --upgrade certifi" can also resolve this issue in some cases.
```

### More Information.

I tried the following commands:
```
./Install\ Certificates.command
```

```
/Library/Frameworks/Python.framework/Versions/3.10/bin/python3 -m pip install --upgrade certifi
```

```
/Applications/Python\ 3.12/Install\ Certificates.command                                       
```

none have fixed the issue

nakajimayoshi, 2024-09-06:
A workaround for this is to go into the tools/detec_python.sh file and delete all the python versions listed except for the one you have installed

closed 2024-09-06 by nakajimayoshi

dobairoland, 2024-09-09:
Hi @nakajimayoshi. And now you use 3.12?

The path to Install Certificates contains 3.12 which looks weird to me.

TAKETODAY, 2025-09-24:
I encountered this issue today as well.

MacOS

I resolved this issue.

idf_tools.py/download()

```python
if site in url:
    ctx = ssl.create_default_context()
    ctx.load_default_certs()
    if platform.system().lower() == 'darwin':
        import certifi
        ctx.load_verify_locations(
            cafile=os.path.relpath(certifi.where()),
            capath=None,
            cadata=cert)
    break
```

https://stackoverflow.com/questions/52805115/certificate-verify-failed-unable-to-get-local-issuer-certificate