espressif/esp-idf #14521: ESP32-C5-DevKitC-1 iperf says 2G is started as default | band set 2G - scan not working - switching to 5G - scan possible but 2.4GHz Hotspots -  5G not working (swapped)

ESP32DE, 2024-09-06:
[image]

( band 2) should been 2.4 GHz, no 2.4 GHz AP found but there are 2.4 GHz AP installed

[image]

( band 5 ) should been 5.8 GHz, scans 2.4 GHz AP's which are installed but no 5.8 GHz AP but there also 5.8 GHz AP installed

[image]

( fresh start should be standard band 2 )
setup AP -> ok but ( but channel is from 5G "112")

[image]

(  band 5 )
after restart set to 5G
setup AP - > ok ( but channel is from 2G "13")
setup a server for test -> ok but  
FYI: iperf -s not working, ports are allways closed after setup the server

[image]

( 2G has no channel 112 - looks like things swapped ( 5G not ready ? ) 

[image]

ESP-IDF 
[image]

lozaning, 2024-09-07:
I can replicate this on the units i've got. the scan results are flipped for the bands. When band 2 is i set i get all 5ghz networks, when band 5 is set I get all 2.4Ghz networks. Both scans work fine, they just return the wrong results for me. 

1. turn unit on and connect over UART
2. unit claims to be set to 2ghz by default
3. start scan
4. see that all returned networks are known 5Ghz networks, and channels returned are for 5Ghz networks
5. set band to 2
6. start scan
7. see the same networks returned as in step 4, all known 5Ghz networks
8. set band to 5
9. start scan
10. see that all returned networks are for known 2.4Ghz networks, and reported channels are all 2.4Ghz channels
11. start scan with channel set to one in the 5Ghz range with the band still set to 5
12. see networks that are all 2.4ghz networks again
[image]

hansw123, 2024-09-12:
@lozaning @ESP32DE 
can you provide the idf commit ,in master iperf ,we don.t have  band command
and in our code band 2 means 5g ,band 1 means 2.4g
[image]

hansw123, 2024-09-12:
@lozaning @ESP32DE 

by the way can your provide the iperf code

ESP32DE, 2024-09-12:
@hansw123 
thank you for your reply.

am confused now :) 
" band 2 means 5g ,band 1 means 2.4g"

where we get the info to use **band 2 for 2.4G** and **band 5 for 5.8G** ..

after your response i can confirm that band 1 works for 2.4G and list 2.4G AP 
also band 2 works for 5.8G and list 5.8G AP  and can be found 

[image]

i can also confirm that band 3 works for a scan that list 2.4G **and** 5.8G
[image]

btw there is also 
band 5
band 6
band 7
band 9
band 10
..
not sure perhabs different sub modes for each band. 
cause this is the factory firmware which comes with the ESP32-C5-DevKitC-1 
we have no source code for this. it is also different from the [master code example](https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/examples/wifi/iperf/main/iperf_example_main.c#L128-L136)
 
mastercode main.c 
does not have ESP32-C5 5G band 

[image]

bootlog ESP32-C5-DevKitC-1

[image]

idf iperf commit:  **ESP-IDF v5.4-dev-624-g3d167a46ff-dirty**

have also a look to the [memory leak ](https://github.com/espressif/esp-idf/issues/14540)

ESP32DE, 2024-09-12:
> ![image](https://private-user-images.githubusercontent.com/73984471/366704290-ef681240-8c73-46cb-8071-8a8fea5a6962.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxODQxODQsIm5iZiI6MTcyNjE4Mzg4NCwicGF0aCI6Ii83Mzk4NDQ3MS8zNjY3MDQyOTAtZWY2ODEyNDAtOGM3My00NmNiLTgwNzEtOGE4ZmVhNWE2OTYyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEyVDIzMzEyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY4ZGViZTkzZTQ5MWY4OWRlYThhZGU4YjUxODM1NjAzYWIzNmY5OWE3YzViYzY0MTZhZDJmZTA3YjU2YWY3MTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Rlz-w4T9-QttVaHODy4_ykIF8OEjhsx0T6WHqQIawbc)

[got it](https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/components/esp_wifi/include/esp_wifi_types_generic.h#L294-L306) - and noted in the master. i will try to switch to master for testing things with the ESP32-C5 

note: in the ESP-IDF V5.4-dev there is no **wifi_band_mode_t** but there is **[wifi_band_t](https://github.com/espressif/esp-idf/blob/0436d4981593e349f82c49007c79f69be8c3cec7/components/esp_wifi/include/esp_wifi_types_generic.h#L1063-L1068)**

your reply helped for this - i noted few missings in the ESP-IDF 5.4-dev and it also not compile successful the iperf example from ESP-IDF V5.4-dev  so i did not switch to 5.4-dev for this yet - i work with the V5.3-dirty but i will give the Masterbrunch  a try for ESP32-C5 now
  
have to test also server / client  again - this did not work / port was closed after opening. 

thank you for your reply!

ESP32DE, 2024-09-13:
@hansw123 

test with 5G, setup as AP and setup as simple (tcp) server - default port - no success

[image]

test with 2G, setup as AP and setup as simple (tcp) server - default port - no success

[image]

plz have also a note to the [memory leak issue](https://github.com/espressif/esp-idf/issues/14540)

hansw123, 2024-09-13:
> @hansw123
> 
> test with 5G, setup as AP and setup as simple (tcp) server - default port - no success
> 
> ![grafik](https://private-user-images.githubusercontent.com/16070445/367105807-b5114661-30cc-46f7-8486-a42d247260aa.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYyMTE5ODEsIm5iZiI6MTcyNjIxMTY4MSwicGF0aCI6Ii8xNjA3MDQ0NS8zNjcxMDU4MDctYjUxMTQ2NjEtMzBjYy00NmY3LTg0ODYtYTQyZDI0NzI2MGFhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEzVDA3MTQ0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWVmMjdkMTBmOWNmODlmNDI4N2Y4NzZlNTVjYzI0MzM2ZjJjNjBiNTcyNWE5ZjljN2U5YWIxNTRhYWNjYWFhZDkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MQAdC1eoUpEEUarr3LBjdIuG3_Zr-gw8XcrS1qEMlIQ)
> 
> test with 2G, setup as AP and setup as simple (tcp) server - default port - no success
> 
> ![grafik](https://private-user-images.githubusercontent.com/16070445/367106212-2947aa69-cd60-440a-9bdd-7f171ad69b3a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYyMTE5ODEsIm5iZiI6MTcyNjIxMTY4MSwicGF0aCI6Ii8xNjA3MDQ0NS8zNjcxMDYyMTItMjk0N2FhNjktY2Q2MC00NDBhLTliZGQtN2YxNzFhZDY5YjNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEzVDA3MTQ0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkZWM3NGZlMzE1YzkwYzNjM2Y3NmY4OGE5NDRkMDllYWExNjBiNzkxOGI2YjAyNjA5ZjgxMGVlNDMyZmY5YjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Eywk2nlWcOWYKIqwl1_ePFrAFrYyQqBWaJ064rk34-o)
> 
> edit: ESP-IDF 5.4 does not have "wifi_band_mode_t" - so i do not switch in iperf any further mode only the simple standard iperf -s i did try also to connect ESP32-C5 as a STA to an HOME AP and then config it as server, same result.
> 
> plz have also a note to the [memory leak issue](https://github.com/espressif/esp-idf/issues/14540)

when the device set as iperf server,it shoule have  other device to connect socket with it in few seconds, if not the server will close auto

ESP32DE, 2024-09-13:
@hansw123 
thank you for reply

5.8G ( band 2) there is no join to AP possible from the second ESP32-C5 ( client )
there are different memory free ( leaks ) there. 
should not been different on start and same cmd also same firmware.

1 usb connected
2 putty start and connect 
3 restart
4 free ( left side:  | right side:  )
5 version ( see picture ) 
6 free ( left side: | right side:  )
7 left side set band 2 | right side set band 2
8 free ( left side: | right side: ) 
9 left side set AP blobfish5g 1234567890 | right side set STA  blobfish5g 1234567890
10 join AP from right side not possible
11 restart
12 free
13 band 1
14 free ( left side: | right side: )
15 restart
16 free ( left side: | right side: )

# steps 1 .. 8

[image]

# steps 9 .. 10 

[image]

# steps 12 .. 14

[image]

# step 16

[image]

ESP32DE, 2024-09-13:
@hansw123 

more steps

- restart
- free
- band 2
- free
- left side set AP blobfish5g 1234567890 | right side scan

no AP found.

the same procedure visa vice .. no better result

[image]

ESP32DE, 2024-09-13:
further steps

"switch on" the Home AP and i can scan the 5G Home AP, but not the ESP32-C5 5G Modul AP

[image]

ESP32DE, 2024-09-13:
The Factory FW is not ready for 5G AP Mode between SoC's.

ESP32DE, 2024-09-13:
further steps 

using a 3rd ESP32-C5 and scan the band 2 

- connect usb
- connect putty
- restart
- free
- version
- band 2
- scan

[image]

ESP32DE, 2024-09-13:
last but not least

# setup a further ESP32-C5 band 5 5GHz AP mode

- usb connect
- putty connect
- restart
- band 2
- mid set AP blobfish5g_mid 1234567890

[image]

# scan with phone

[image]

i can but it makes no sense to setup more
i can not confirm that any factory things fails cause Hardware
i think it is more a Software thing - factory fw is not ready yet at this time point with ESP-IDF V5.4
also the chip info is not right, i think the factory fw does not support this yet.

iperf> version
IDF Version:v5.4-dev-624-g3d167a46ff-dirty
Chip info:
        model:Unknown
        cores:1
        feature:/802.11bgn/BLE/External-Flash:4 MB
        revision number:0
iperf>
 

i will try the master branch next time.
the factory fw should been updated! there is no way to work with this yet

best wishes

ESP32DE, 2024-09-13:
after using masterbranch mixed with parts newer and older [lib](https://github.com/espressif/esp32-wifi-lib) , ESP-IDF V 5.3 dirty,  ESP-IDF V 5.4 dirty and factory [FW](https://en.wikipedia.org/wiki/Firmware) [RE](https://en.wikipedia.org/wiki/Reverse_engineering) and building a [CFW](https://en.wikipedia.org/wiki/Custom_firmware)
 
i have this result now:

but i must say this: 
at this time point -  the band 2 ( 5Ghz ) mode and setup the ESP32-C5 itself as AP and CLI and connect them together
**_does not working_** in the 5GHz ( band 2 ) ,  the AP mode does not support a connect from a client - that is fact at time point yet. DOT. i think it's simple not supported yet in the 5GHz band.

# 1st steps

- set up a Home 5Ghz AP
- set up 2 ESP32-C5 Kit on band 2 ( 5GHz ) and connect to the Home 5GHz AP
- prepair the 2 ESP32-C5 kit 
  - one as server ( 192.168.155.86 ) 
  - one as client ( 192.168.155.39 ) 

[image]

# next steps
 
- press enter , start the server , start the client and generate net traffic

[image]

# last but not least

-  very limited running with the help of external components (1Gbit AP Wifi) 

[image]

It's such a pity and sad that once again no one from espressif ( @igrr @costaud ) itself gets in touch. It would be more helpful to give some information that it just doesn't work yet, instead some people run into the wall again. 

I'm closing this now because I actually expect people to take it seriously and to heart when products are delivered. 
No, they prefer to remain silent again. Unfortunately, this makes strong marketing the weakest link in the presentation.

hansw123, 2024-09-14:
@ESP32DE 
very very sorry for late reply, Thank you very much for your feedback on the problem, your test steps are the most detailed and conscientious I've ever seen!I can feel your passion for espressif products!
But since I'm based in China, there might be a bit of a time difference.I apologize for not getting back to you in time！

hansw123, 2024-09-14:
@ESP32DE 
This is an issue we always deal with at a very high priority. It's safe to assume now that the firmware you got is relatively early and does have some issues.

hansw123, 2024-09-14:
@ESP32DE 
maybe you can try 34813ecd6ce44bc6c42274ded2f9043434384645 this commit 
and use iperf example retest.
Some of the issues you mentioned have been found internally and fixed on this release

ESP32DE, 2024-09-14:
@hansw123 
just read in .. and do fast reply 
thank you for your effort and service
JLYK there is a sync in the pipe  https://github.com/espressif/esp-idf/issues/14540#issuecomment-2350814177
i come back to this in few mins..

ESP32DE, 2024-09-14:
> @ESP32DE maybe you can try [34813ec](https://github.com/espressif/esp-idf/commit/34813ecd6ce44bc6c42274ded2f9043434384645) this commit and use iperf example retest. Some of the issues you mentioned have been found internally and fixed on this release

thank you @hansw123 
i give it a try and let you asap know
i will compare it with the sync in https://github.com/espressif/esp-idf/issues/14540#issuecomment-2350814177 

i am not sure does the [34813ec](https://github.com/espressif/esp-idf/commit/34813ecd6ce44bc6c42274ded2f9043434384645) explizit helps cause target lcd, mipi, mcpwm and rmt ( dsi ) 
[image]

but i will investigate in this code - 
thank you very much for the hint with band 1 and band 2 -
 i did not checked this in the master directly by searching 
but i was missing some types ..which was seen symbolic in the Factory FW 

will update this here but for now
wish you also a happy mid-autumn (中秋节) festival in advance if we not read/write next days
( yes its more as passion -  )
best wishes

hansw123, 2024-09-14:
@ESP32DE 
My commit point for this was before #14540 (comment). Also chose this point to make sure that some of the wifi fixes have been merged.

ESP32DE, 2024-09-15:
@hansw123 
checked and [figured it out](https://github.com/espressif/esp-idf/issues/14540#issuecomment-2351237129)
one bit is different - you see ? :)
 
[image]

thank you very much for your efforts and service to bring me to the right spot.
i just test all now again - 
i will investigate where the chip infos is gone
should come from sdkconfig.h usually - i will check.
keep this here in update state for any news.

;-)

closed 2025-01-24 by MaxwellAlan