espressif/esp-idf #14522: Getting esp serial number in code

damian-kurek-wizzdev, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I got a request from the factory to provide them esp32c6 serial number via REST API. The serial number is taken from the QR code that they scan. Message from them:

> This number is scanned from the QR code from ESP WIFI modular which is displayed on the computer；
The code looks like this:
[image]
Can I somehow get this code into the firmware? The part in the red box is a serial number. I had a look into the esp-idf docs and couldn't find anything like that

atanisoft, 2024-09-07:
That looks like the MAC address of the ESP32-C6 SoC. You should be able to retrieve if via:
```
uint8_t mac_addr[6];
esp_read_mac(mac_addr, ESP_MAC_WIFI_SOFTAP)
```
The `ESP_MAC_WIFI_SOFTAP` parameter determines which MAC will be returned, there are a few options to pick from.

arosyc, 2024-09-09:
You can confirm QR code discription in part 2.3:
https://www.espressif.com/sites/default/files/documentation/espressif_module_packaging_information_en.pdf

So, I think you can use MAC ID as serial number. Because it is unique.

damian-kurek-wizzdev, 2024-09-09:
@arosyc ok.
Can I get digits 1 to 6 from esp32?
[image]

arosyc, 2025-03-07:
@damian-kurek-wizzdev  I'm afraid not. The information from 1 to 6 cannot be obtained through the chip. This is because it was laser-marked on the module externally when produced by Espressif, and it has no relation to the chip itself.

closed 2025-03-07 by Alvin1Zhang