espressif/esp-idf #14524: Memory Allocation Issue (ESP RainMaker + ESP-WHO Face Detection + BLE NimBLE)

NEHAL608, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Espressif SoC revision.

ESP32-S3 WROOM2 with 32MB

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

I'm working on a project with the ESP32-S3 where I need to integrate ESP RainMaker, BLE NimBLE, and the Face Detection Web API from ESP-WHO. When I run each API individually, or even combine ESP RainMaker and BLE NimBLE, everything works perfectly fine. However, when I try to combine all three (ESP RainMaker + BLE + Face Detection), I run into memory allocation failures.
I have already tried various memory optimizations.   

Has anyone faced similar issues or have suggestions for further memory optimizations to run all three components successfully on the ESP32-S3? Any advice or alternative approaches would be greatly appreciated!

Thanks in advance!

### What is the actual behavior?

none

### Steps to reproduce.

1 idf.py build 
2 idf.py flash 

### Debug Logs.

```plain
I (5873) app_wifi: Connected with IP Address:192.168.0.108
I (5873) esp_netif_handlers: sta ip: 192.168.0.108, mask: 255.255.255.0, gw: 192.168.0.1
I (5873) main_task: Returned from app_main()
I (5873) esp_rmaker_cmd_resp: Enabling Command-Response Module.
I (5883) esp_rmaker_node: Node attribute cmd-resp created
I (5893) esp_rmaker_local: Enabling Local Control
I (5893) esp_rmaker_local: Local Control Service Enabled
I (5903) esp_rmaker_local: Starting ESP Local control with HTTP Transport and security version: 1
I (5913) esp_https_server: Starting server
I (5923) esp_https_server: Server listening on port 8080
I (5923) esp_rmaker_local: esp_local_ctrl service started with name : 348518A1BDC0
I (5933) esp_mqtt_glue: Connecting to a1p72mufdu6064-ats.iot.us-east-1.amazonaws.com
I (5943) esp_rmaker_core: Waiting for MQTT connection
I (6133) ioexpander: fa
I (6133) ADC: ADC1 Channel[1] Raw Data: 4095
I (6133) ADC: ADC1 Channel[1] Cali Voltage: 3076 mV
I (6133) ADC: Resistance: 274642.84 Ohms, ambient_Temperature: -36.82°C

I (6533) esp-x509-crt-bundle: Certificate validated
I (8153) ADC: Raw Data: 52 MAIN MOTOR Cali Voltage: 44 mV
I (8193) esp_mqtt_glue: MQTT Connected
I (8193) esp_rmaker_cmd_resp: Checking for pending commands.
I (8193) esp_rmaker_node_config: Generated Node config of length 2180
I (8203) esp_rmaker_param: Params MQTT Init done.
I (8203) esp_rmaker_param: Reporting params (init): {"MotorFanLight":{"Motor Speed":0,"Fan Speed":0,"Light Brightness":0},"read Sensor":{"Name":"read Sensor","Temperature":-36.82250,"battery Sensor":0.00000},"Time set":{"Hour":0,"Minute":0,"Second":0},"Time":{"TZ":"Asia/Kolkata","TZ-POSIX":"IST-5:30"},"Schedule":{"Schedules":[]},"Scenes":{"Scenes":[]},"Local Control":{"POP":"90aef653","Type":1}}
I (8243) esp_rmaker_param: Reporting Time Series Data for read Sensor.Temperature
I (8553) esp_rmaker_param: Reporting Time Series Data for read Sensor.Temperature
I (8553) esp_rmaker_param: Reporting params: {"read Sensor":{"Temperature":-36.82250}}
I (8563) app_main: Provisioning completed
I (8563) BLE_INIT: BT controller compile version [ba3b3e3]
I (8573) BLE_INIT: Bluetooth MAC: 34:85:18:a1:bd:c2
E (8573) BLE_INIT: Malloc failed
assert emi.c 164, param 00000000 00008400
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40006fcb  PS      : 0x00060534  A0      : 0x8000a3d8  A1      : 0x3fcf7340
0x40006fcb: r_assert_param in ROM

A2      : 0x00000000  A3      : 0x00008400  A4      : 0x3ff196ce  A5      : 0x000000a4
A6      : 0x3fceff98  A7      : 0x0000cdcd  A8      : 0x00000001  A9      : 0x3fcef9e4
A10     : 0x3fcef9e4  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00008400
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000005  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x40006fc4  LEND    : 0x40006fcb  LCOUNT  : 0x00000000
0x40006fc4: r_assert_param in ROM
0x40006fcb: r_assert_param in ROM

Backtrace: 0x40006fc8:0x3fcf7340 0x4000a3d5:0x3fcf7360 0x40007303:0x3fcf7380 0x4206da03:0x3fcf73a0 0x4206e0fe:0x3fcf73e0 0x420752ef:0x3fcf7420 0x42075a11:0x3fcf7450 0x4206b018:0x3fcf7470 0x403785d0:0x3fcf7490 0x403850aa:0x3fcf74c0
0x40006fc8: r_assert_param in ROM
0x4000a3d5: r_emi_get_mem_addr_by_offset in ROM
0x40007303: r_ble_util_buf_rx_alloc in ROM
0x4206da03: r_lld_core_init at ??:?
0x4206e0fe: r_lld_init at ??:?
0x420752ef: r_rwble_init at ??:?
0x42075a11: r_rwip_reset at ??:?
0x4206b018: btdm_controller_on_reset at ??:?
0x403785d0: btdm_controller_task at ??:?
0x403850aa: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.3/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

Core  1 register dump:
PC      : 0x403801ce  PS      : 0x00060334  A0      : 0x82005ffd  A1      : 0x3fcbd520
A2      : 0x00000000  A3      : 0x0000abab  A4      : 0x80385456  A5      : 0x3fcbe240
A6      : 0x00060023  A7      : 0x00000000  A8      : 0x8209d20a  A9      : 0x3fcbd4e0
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x80385456  A13     : 0x3fcc7db0
A14     : 0x00060023  A15     : 0x3fcbd6f4  SAR     : 0x00000000  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace: 0x403801cb:0x3fcbd520 0x42005ffa:0x3fcbd540 0x403867a5:0x3fcbd560 0x403850aa:0x3fcbd580

ELF file SHA256: af711645f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40376088
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
load:0x3fce2810,len:0x178c
load:0x403c8700,len:0x4
load:0x403c8704,len:0xcb8
load:0x403cb700,len:0x2df4
entry 0x403c8914
```

### More Information.

[image]
[sdkconfig.zip](https://github.com/user-attachments/files/16912063/sdkconfig.zip)
[image]

Sandra-lol, 2024-09-09:
Hi, could you try initiate other app, before initializing the esp-dl models?

NEHAL608, 2024-09-09:
No, now I initialized ESP RainMaker and successfully completed the provisioning. After that, I initialized BLE, and only then did I initialize the ESP-DL model."  i got this error 
[image]