espressif/esp-idf #14525: warning: variable 'other_core_owner_id' set but not used

xmow49, 2024-09-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect no warns build from the sdk.

### What is the actual behavior?

When I build my projet on a esp32s3 I have this warning:
```
[1061/1092] Building C object esp-idf/lidar/CMakeFiles/__idf_lidar.dir/ld19/lidar_ld19.c.obj
In file included from /home/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:49,
                 from /home/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/portable.h:57,
                 from /home/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:69,
                 from /home/esp/esp-idf/components/esp_driver_uart/include/driver/uart.h:12,
                 from /home/esp/esp-pami/esp-qs/our_uart/our_uart.h:11,
                 from /home/esp/esp-pami/src/lidar/ld19/lidar_ld19.c:23:
/home/esp/esp-idf/components/esp_hw_support/include/spinlock.h: In function 'spinlock_acquire':
/home/esp/esp-idf/components/esp_hw_support/include/spinlock.h:78:29: warning: variable 'other_core_owner_id' set but not used [-Wunused-but-set-variable]
   78 |     uint32_t core_owner_id, other_core_owner_id;
      |                             ^~~~~~~~~~~~~~~~~~~
/home/esp/esp-idf/components/esp_hw_support/include/spinlock.h: In function 'spinlock_release':
/home/esp/esp-idf/components/esp_hw_support/include/spinlock.h:176:14: warning: variable 'core_owner_id' set but not used [-Wunused-but-set-variable]
  176 |     uint32_t core_owner_id;
      |              ^~~~~~~~~~~~~
[1091/1092] Generating binary image from built executable
esptool.py v4.8.dev5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
```

### Steps to reproduce.

I'm not able to reproduce it outside of my project.
I don't understand why there is this warn.

SoucheSouche, 2024-09-09:
Hi @xmow49,
Thank you for reporting the issue!
We found out about it as well a few weeks back and it has been fixed.
The code fix was already synced with GH master branch (0d140f38ea38deff6b6edf63616e7725b0c7a269).

closed 2024-09-09 by SoucheSouche

xmow49, 2024-09-11:
I will wait the next release (maybe the 5.3.2 ?) 
I'm in no hurry.