espressif/esp-idf #14527: Discussion: Is there a way to access the MPI functions in the ROM?

ricmoo, 2024-09-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I found [the following](https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/components/mbedtls/port/mbedtls_rom/mbedtls_rom_osi.c) file while porting the hardware MPI support to the Stub and was wondering if there is a way to access these functions directly from the ROM (including the hardware accelerated `mbedtls_mpi_exp_mod`).

Thanks.

nopnop2002, 2024-09-17:
I'm not sure if it meets your requirements, but I was able to use mbedtls_aes_init() inside mbedtls_rom_funcs_table.

https://github.com/nopnop2002/esp-idf-mbedtls/tree/main/aes_crypt

AdityaHPatwardhan, 2024-09-18:
Hi @ricmoo,

At the moment only the esp32c2 SoC contains some functions from mbedTLS in its ROM.
These functions are the pure software implementation from the mbedTLS 2.16.9 version. However, They shall not provide the hardware acceleration. For hardware acceleration you need to involve the port layer from mbedtls.

The list of mbedtls functions that are present in the ROM of esp32c2 SoC can be found out at https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/components/esp_rom/esp32c2/ld/esp32c2.rom.mbedtls.ld

ricmoo, 2024-09-18:
I see. I was hoping to remove some functions from my code (like add, mul, etc.) in favour of using the ROM versions, but there isn’t really any function on the list I use for prime number generation. :)

Thanks! :)

closed 2024-09-18 by ricmoo