espressif/esp-idf #14530: Encountered strange issues while debugging the serial port with the C6 ULP core

Tinyu-Zhao, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I encountered a strange problem when using the ULP core to debug with the serial port, the same code, but the placement is different, there is no other information in the middle, one line is indeed normal output information 'safe_power 125 ccc', the other line is garbled `saæ}ÿpo W.
ª`, and the result of multiple attempts is the same.

[image]

SoucheSouche, 2024-09-09:
Hi @Tinyu-Zhao,
This is strange indeed. We will take a look and come back to you.

Tinyu-Zhao, 2024-09-09:
> Hi @Tinyu-Zhao, This is strange indeed. We will take a look and come back to you.

Thank you and I will be happy to provide more information.
Current IDF Version is :ESP-IDF v5.4-dev-2620-g6673376297-dirty

sudeep-mohanty, 2024-09-09:
Hi @Tinyu-Zhao,
What you are observing is likely an issue with the lp_uart driver which does not wait for the UART Tx FIFO to flush out before printing the next string/character. Could you please apply the following patch and let us know if it resolves the issue?

[lp_uart.patch](https://github.com/user-attachments/files/16930554/lp_uart.patch)

Tinyu-Zhao, 2024-09-10:
> Hi @Tinyu-Zhao, What you are observing is likely an issue with the lp_uart driver which does not wait for the UART Tx FIFO to flush out before printing the next string/character. Could you please [...]

Thank you for the quick response, it worked properly!
[image]

sudeep-mohanty, 2024-09-10:
Thanks for the confirmation @Tinyu-Zhao. I shall do some more tests and make a change.

closed 2024-09-21 by espressif-bot