espressif/esp-idf #14531: SoftAp mode : WiFi ping delay is high and unstable

ShengjieCh, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.2

### Espressif SoC revision.

ESP32C6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

In an interference-free environment, the ping delay when WiFi is used as a softAP is low and stable.

### What is the actual behavior?

the ping delay is high and unstable.

### Steps to reproduce.

Anyway, I tested the ping integrated in [default esp-idf iperf example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/iperf) (native, i.e. non-esp-hosted) and found below result over the air test:

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0xe (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x1974
load:0x4086c110,len:0x1018
load:0x4086e610,len:0x2f68
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4-dev-1915-g15514919c9-dirty 2nd stage bootloader
I (24) boot: compile time Aug 21 2024 11:52:48
I (25) boot: chip revision: v0.0
I (29) qio_mode: Enabling default flash chip QIO
I (34) boot.esp32c6: SPI Speed      : 80MHz
I (39) boot.esp32c6: SPI Mode       : QIO
I (43) boot.esp32c6: SPI Flash Size : 2MB
I (48) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
[... 189 lines cut ...]
64 bytes from 192.168.4.2: icmp_seq=20 ttl=64 time=2 ms
64 bytes from 192.168.4.2: icmp_seq=21 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=22 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=23 ttl=64 time=2 ms
64 bytes from 192.168.4.2: icmp_seq=24 ttl=64 time=1 ms
64 bytes from 192.168.4.2: icmp_seq=25 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=26 ttl=64 time=11 ms
64 bytes from 192.168.4.2: icmp_seq=27 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=28 ttl=64 time=18 ms
64 bytes from 192.168.4.2: icmp_seq=29 ttl=64 time=5 ms
64 bytes from 192.168.4.2: icmp_seq=30 ttl=64 time=8 ms
64 bytes from 192.168.4.2: icmp_seq=31 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=32 ttl=64 time=10 ms
64 bytes from 192.168.4.2: icmp_seq=33 ttl=64 time=35 ms
64 bytes from 192.168.4.2: icmp_seq=34 ttl=64 time=59 ms
64 bytes from 192.168.4.2: icmp_seq=35 ttl=64 time=67 ms
64 bytes from 192.168.4.2: icmp_seq=36 ttl=64 time=107 ms
64 bytes from 192.168.4.2: icmp_seq=37 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=38 ttl=64 time=52 ms
64 bytes from 192.168.4.2: icmp_seq=39 ttl=64 time=3 ms
64 bytes from 192.168.4.2: icmp_seq=40 ttl=64 time=4 ms

--- 192.168.4.2 ping statistics ---
40 packets transmitted, 40 received, 0% packet loss, time 787ms
```

jyc6, 2024-09-13:
我也遇到了这个问题，ap模式还总出现断的情况；

mhdong, 2024-09-14:
Hi @ShengjieCh 
I have already tried to reproduce this issue locally. I was able to reproduce the same phenomenon as you observed. Through packet capture and log analysis, it can be concluded that when the ping latency is high, such as over 50ms, it typically results in BA loss. I also conducted the same test in a shielded box environment, where the issue of high ping latency did not occur.
[softap_mode_ping_delay.zip](https://github.com/user-attachments/files/17001522/softap_mode_ping_delay.zip)
STA mac: wlan.addr == 60:55:f9:f6:04:24
AP mac: wlan.addr == 60:55:f9:f6:2e:d1
SSID: testmhd
password: 12345678
We will continue to analyze the cause of BA loss. If it is due to environmental interference, we will consider whether to adjust the BA rate control algorithm to enhance the robustness of BA packets.

mhdong, 2024-09-14:
hi @jyc6 
你好，方便的话请提供一下IDF的版本和复现环境。如果能提供复现的log和抓包，这有利于我们快速的分析问题的原因。

AxelLin, 2024-10-13:
> I was able to reproduce the same phenomenon as you observed. We will continue to analyze the cause of BA loss. 

@mhdong Any progress for this issue?

ShengjieCh, 2024-10-18:
Ping latency test results of chip B：https://pan-sec.djicorp.com/s/H6tq8Tj4xWxDYfW

AxelLin, 2024-11-02:
@mhdong 
Did you check the wireshark log in https://github.com/espressif/esp-idf/issues/14531#issuecomment-2421630444 ?
Do you have any update for this issue?

AxelLin, 2024-11-19:
@mhdong 
There is no update for 2 Months since you replied you can reproduce this issue on Sep 14. https://github.com/espressif/esp-idf/issues/14531#issuecomment-2350902793.
Do you have any progress for the fix?

mhdong, 2024-11-19:
[image]
[image]
hi @ShengjieCh  @AxelLin 
Based on the analysis of the provided packet captures and logs, it has been confirmed that the high ping latency is caused by the STA entering sleep mode.

AxelLin, 2024-11-19:
@mhdong 
Does your test show the same reason for this issue (since you can reproduce it locally).

mhdong, 2024-11-19:
hi @AxelLin 
The situation I reproduced may not be the same as the customer's. In the shielding box, I couldn't reproduce the issue, but in a noisy environment, there is an increase in ping latency. From the packet capture, when the latency increases, there are a large number of other packets being transmitted between ping request and ping reply.

closed 2024-11-21 by Sherry616