espressif/esp-idf #14532: Cannot build Component example if Component directory has different name than published Component name

abobija, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2440-g0de2912f91

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected to be able to build the Component example even if the local Component directory name differs from the published Component's name.

Local Component path: `~/repos/esp-idf-rc522` (cloned from [here](https://github.com/abobija/esp-idf-rc522))
Component name on Component Registry: [`abobija/rc522`](https://components.espressif.com/components/abobija/rc522)

### What is the actual behavior?

Instead, it throws me an error:

```
The component abobija__rc522 could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
```

### Steps to reproduce.

This throws an error that component cannot be found:

1. `git clone git@github.com:abobija/esp-idf-rc522.git`
2. `cd esp-idf-rc522/examples/basic`
3. `idf.py build`

But if we clone the repo into local dir named `rc522` (because that is the name of [published Component](https://components.espressif.com/components/abobija/rc522)) then it works:

1. `git clone git@github.com:abobija/esp-idf-rc522.git rc522`
2. `cd rc522/examples/basic`
3. `idf.py build`

### Build or installation Logs.

```plain
idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/abobija/repos/esp-idf-rc522/examples/basic/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/abobija/.espressif/python_env/idf5.3_py3.12_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/abobija/repos/esp-idf-rc522/examples/basic"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/home/abobija/repos/esp-idf-rc522/examples/basic/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.43.0")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/abobija/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/abobija/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/abobija/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
Using component placed at /home/abobija/repos/esp-idf-rc522 for dependency "abobija/rc522", specified in /home/abobija/repos/esp-idf-rc522/examples/basic/main/idf_component.yml
Processing 2 dependencies:
[1/2] abobija/rc522 (2.6.1) (/home/abobija/repos/esp-idf-rc522)
[2/2] idf (5.3.0)
CMake Error at /home/abobija/repos/esp-idf/tools/cmake/build.cmake:268 (message):
  Failed to resolve component 'abobija__rc522'.
Call Stack (most recent call first):
  /home/abobija/repos/esp-idf/tools/cmake/build.cmake:310 (__build_resolve_and_add_req)
  /home/abobija/repos/esp-idf/tools/cmake/build.cmake:599 (__build_expand_requirements)
  /home/abobija/repos/esp-idf/tools/cmake/project.cmake:710 (idf_build_process)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
HINT: The component abobija__rc522 could not be found. This could be because: component name was misspelled, the component was not added to the build, the component has been moved to the IDF component manager or has been removed and refactored into some other component.
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details about moved components.
Refer to the build-system guide for more details about how components are found and included in the build.
cmake failed with exit code 1, output of the command is in the /home/abobija/repos/esp-idf-rc522/examples/basic/build/log/idf_py_stderr_output_566422 and /home/abobija/repos/esp-idf-rc522/examples/basic/build/log/idf_py_stdout_output_566422
```

### More Information.

- I don't like the idea of publishing a component like `abobija/esp-idf-rc522` to match the repository name.
- I also don't like the idea of renaming the repository to `rc522` to match published Component name because of side effects.
- I can write instructions for local development to clone the repo into the local `rc522` directory using `git clone git@github.com:abobija/esp-idf-rc522.git rc522`, but I think that would be a bit messy for contributors.

Is it possible to modify the build tool so that it figures out that it's the same component and doesn't throw an error if the local component directory name is different from the published name?

kumekay, 2024-09-09:
Hello @abobija, thank you for opening the issue, 
you are right, currently ESP-IDF CMake build system uses directory names internally, and there is currently no way to pass the real component name from the [Component Manager](https://github.com/espressif/idf-component-manager) to the build system. I think it can be changed in the future ESP-IDF releases, but not for existing ones.

abobija, 2024-09-09:
Thank you for your response. I sincerely hope for prioritization of this issue as it will allow flexibility in publishing components 🙏🏼