espressif/esp-idf #14533: Kernel panic when using cJSON in mqtt

Gyunamchoi, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

Create JSON structure -> Convert to string -> Proceed with mqtt publish

### What is the actual behavior?

During the AWS IoT OTA process, an error occurs in malloc when calling cJSON library processes such as cJSON_CreateObject(); or cJSON_Print().

### Steps to reproduce.

1. System boot
2. Network, mqtt initialization
3. mqtt subscribe (OTA related topic)
4. MQTT_EVENT_DATA event occurs
5. Topic parsing process in progress
6. Check $aws/things/(device_id)/jobs/execution/status
7. $aws/things/(device_id)/streams/(device_id)/get/json topic, build data JSON and send
8. Kernel panic occurs when building JSON here

### Debug Logs.

```plain
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4008d096  PS      : 0x00060233  A0      : 0x8008ca34  A1      : 0x3ffc73f0  
0x4008d096: remove_free_block at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:332
 (inlined by) block_locate_free at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:567
 (inlined by) tlsf_malloc at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:1015

A2      : 0x3ffd0a98  A3      : 0x0000002c  A4      : 0x3ffd2e8c  A5      : 0x3ffd09d8  
A6      : 0x3ffd129c  A7      : 0x3ffb4fbc  A8      : 0x0000000b  A9      : 0xffffffff  
A10     : 0x00000002  A11     : 0xffffffff  A12     : 0x00000464  A13     : 0x3ffb4fe0  
A14     : 0x3f404998  A15     : 0x0000cdcd  SAR     : 0x0000001f  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x0000000b  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x4008d093:0x3ffc73f0 0x4008ca31:0x3ffc7410 0x40082745:0x3ffc7430 0x40082862:0x3ffc7450 0x4008289d:0x3ffc7480 0x40082646:0x3ffc74a0 0x4008269e:0x3ffc74f0 0x400820e4:0x3ffc7510 0x40082119:0x3ffc7530 0x4008e298:0x3ffc7550 0x4008e2bc:0x3ffc7570 0x400e8389:0x3ffc7590 0x400e6ce7:0x3ffc75c0 0x400e6d85:0x3ffc7600 0x400e7b90:0x3ffc7620 0x400db9cb:0x3ffc7660 0x400dbe2f:0x3ffc7690 0x400dc8a9:0x3ffc76e0 0x400dcba9:0x3ffc7700 0x40149ec9:0x3ffc77a0 0x4014995e:0x3ffc77d0 0x400e599a:0x3ffc7810 0x400e63fc:0x3ffc7830 0x400e6563:0x3ffc7860 0x400e6a60:0x3ffc7890 0x400882c9:0x3ffc78c0
0x4008d093: remove_free_block at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:331 (discriminator 1)
 (inlined by) block_locate_free at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:567 (discriminator 1)
 (inlined by) tlsf_malloc at /Users/odn/esp/esp-idf/components/heap/tlsf/tlsf.c:1015 (discriminator 1)
0x4008ca31: multi_heap_malloc_impl at /Users/odn/esp/esp-idf/components/heap/multi_heap.c:211
0x40082745: aligned_or_unaligned_alloc at /Users/odn/esp/esp-idf/components/heap/heap_caps_base.c:77
0x40082862: heap_caps_aligned_alloc_base at /Users/odn/esp/esp-idf/components/heap/heap_caps_base.c:150
0x4008289d: heap_caps_malloc_base at /Users/odn/esp/esp-idf/components/heap/heap_caps_base.c:170
0x40082646: trace_malloc at /Users/odn/esp/esp-idf/components/heap/include/heap_trace.inc:116
0x4008269e: __wrap_heap_caps_malloc_base at /Users/odn/esp/esp-idf/components/heap/include/heap_trace.inc:178
0x400820e4: heap_caps_malloc at /Users/odn/esp/esp-idf/components/heap/heap_caps.c:84
0x40082119: heap_caps_malloc_default at /Users/odn/esp/esp-idf/components/heap/heap_caps.c:110
0x4008e298: _calloc_r at /Users/odn/esp/esp-idf/components/newlib/heap.c:65
0x4008e2bc: calloc at /Users/odn/esp/esp-idf/components/newlib/heap.c:29 (discriminator 1)
0x400e8389: outbox_enqueue at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/lib/mqtt_outbox.c:44
0x400e6ce7: mqtt_enqueue at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1190 (discriminator 1)
0x400e6d85: mqtt_client_enqueue_publish at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:2038
0x400e7b90: esp_mqtt_client_publish at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:2092
0x400db9cb: mqtt_publish at /Users/odn/workspace/odn_device_rev2/components/mqtt_handle/mqtt_handler.c:201
0x400dbe2f: mqtt_recvtopic_jobs at /Users/odn/workspace/odn_device_rev2/components/mqtt_handle/mqtt_handler.c:586
0x400dc8a9: mqtt_recvprocessing at /Users/odn/workspace/odn_device_rev2/components/mqtt_handle/mqtt_handler.c:402
0x400dcba9: mqtt_eventhandler at /Users/odn/workspace/odn_device_rev2/components/mqtt_handle/mqtt_handler.c:338
0x40149ec9: handler_execute at /Users/odn/esp/esp-idf/components/esp_event/esp_event.c:135
0x4014995e: esp_event_loop_run at /Users/odn/esp/esp-idf/components/esp_event/esp_event.c:584
0x400e599a: esp_mqtt_dispatch_event at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1032
0x400e63fc: deliver_publish at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1099
0x400e6563: mqtt_process_receive at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1377
0x400e6a60: esp_mqtt_task at /Users/odn/esp/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1657
0x400882c9: vPortTaskWrapper at /Users/odn/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134
```

### More Information.

I am implementing OTA using MQTT.
JSON is using the library below.
https://github.com/DaveGamble/cJSON/tree/acc76239bee01d8e9c858ae2cab296704e52d916 Currently, there is no problem with data exchange, subscription, publishing, and other basic operations.

When the MQTT_EVENT_DATA event occurs, the arrived topic is parsed and various operations are performed. At this time, if a JSON file is created, added, or printed, the same operation as the debug screen is performed.

This message occurs when the $aws/things/(device_id)/jobs side for OTA is in progress. It occurs in the part where cJSON_Create, Print, Add, etc. try to secure additional memory with malloc.

The message seems to be related to the malloc function, so I changed the point where free is performed, but it did not work, and the heap memory was checked and entered with more than 170KB remaining just before entering. There was also a solution to add cJSON_InitHooks(NULL); just before entry, but it didn't work.

Current event processing process
1. Event processing handler
void mqtt_eventhandler(void *arg, esp_event_base_t base, int32_t event_id, void *event_data)

2. MQTT_EVENT_DATA processing process
esp_err_t mqtt_recvprocessing(char* topic, char* event_data)

3. $aws/things/(device_id)/jobs processing process
esp_err_t mqtt_recvtopic_jobs(char* remain_topic, char* event_data){
(Internal error)

Please provide interpretation or solution for the message, or if the json handling library is wrong, please recommend an alternative library.

Thank you.

ermacv, 2024-09-10:
Hi! Are you parsing the JSON OTA image? What is the size of this data?

Gyunamchoi, 2024-09-10:
> Hi! Are you parsing the JSON OTA image? What is the size of this data?

OTA images are received through streams, divided into 16 blocks with a block size of 4096 bytes.
References are provided below.
https://docs.aws.amazon.com/iot/latest/developerguide/mqtt-based-file-delivery-in-devices.html

euripedesrocha, 2024-09-17:
Hi @Gyunamchoi I suggest that you enable [heap tracing](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html) and try to identify if there is any heap corruption happening in your code.

chegewara, 2024-09-20:
Be careful when you are using `cJSON_Print`.
It is allocating memory which you have to free explicitly or it will cause memory leak.

Gyunamchoi, 2024-10-01:
> 안녕@Gyunamchoi[](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html)코드에서 힙 손상을 확인하여 [추적을](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/heap_debug.html) 활성화하는 것이 좋습니다.

I turned on heap tracking and checked, and there was 170KB of heap left before the problem occurred.

Gyunamchoi, 2024-10-01:
> Be careful when you are using `cJSON_Print`. It is allocating memory which you have to free explicitly or it will cause memory leak.

Let me ask you again, when you do char* A = cJSON_Print (B);, is the problematic string A or the json structure B?
A is free(A);. B is cJSON_delete(B); to return memory.

chegewara, 2024-10-01:
```
char* A = cJSON_Print (B);
// consume A
...
free(A);
```
`A` is now independent object which exists even after deleting `B` until you `free(A)`.

Aside of this you have to handle `B` as usual.

Gyunamchoi, 2024-10-01:
> ```
> char* A = cJSON_Print (B);
> // consume A
> ...
> free(A);
> ```
> 
> `A` is now independent object which exists even after deleting `B` until you `free(A)`.
> 
> Aside of this you have to handle `B` as usual.

I understand what you're saying.
cJSON_Print is analyzed as allocating memory with malloc in an internal function and returning a char* type address.
In this process, I have a question. For example, in the case of ESP_LOGI(TAG, "string = %s", cJSON_Print(B));, does the result address value of cJSON_Print(B) automatically get destroyed? Or should I put it in a separate variable and free it?

Since there is no garbage collector, I think I have to do this part manually, but I wonder if this structure is okay.

Gyunamchoi, 2024-10-01:
> > > ```
> > > char* A = cJSON_Print (B);
> > > // consume A
> > > ...
> > > free(A);
> > > ```
> > > 
> > > 
> > >     
> > >       
> > >     
> > > 
> > >       
> > >     
> > > 
> > >     
> > >   
> > > `A``B`이제 삭제 후에도 존재하는 독립적인 객체입니다 `free(A)`. 
> > > 이 외에는 `B`평소처럼 처리해야 합니다.
> > 
> > 
> > 말씀하시는 바를 이해합니다. cJSON_Print는 내부 함수에서 malloc으로 메모리를 할당하고 char* 타입 주소를 반환하는 것으로 분석됩니다. 이 과정에서 질문이 있습니다. 예를 들어 ESP_LOGI(TAG, "string = %s", cJSON_Print(B));의 경우 cJSON_Print(B)의 결과 주소 값은 자동으로 파괴되나요? 아니면 별도의 변수에 넣고 해제해야 하나요?
> > 가비지 콜렉터가 없으니 이 부분은 수동으로 해야 할 것 같은데, 이런 구조가 괜찮을지 궁금합니다.
> 
> 이것은 메모리 누수의 예입니다. 별도의 변수에 넣고 나중에 해제해야 합니다.

Thank you very much for your reply.

Actually, the problem was somewhere else and I solved it today.
However, I was able to supplement the firmware according to your advice.

Gyunamchoi, 2024-10-01:
This issue has been resolved and will be closed.

closed 2024-10-01 by Gyunamchoi