espressif/esp-idf #14537: Mesh network nodes cannot ping

yanmartins, 2024-09-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.4

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Nodes must be able to ping each other

### What is the actual behavior?

Nodes can resolve DNS but cannot ping

### Steps to reproduce.

1. Config mesh network
2. Try to ping some address in Nodes

### Debug Logs.

```plain
I (42573) srv_ping: DNS lookup success
I (42583) srv_ping: target_addr.type=0
I (42583) srv_ping: target_addr.u_addr.ip4=x.x.x.x
I (42773) app_main: Heap size -> 8088740
I (42823) mesh: 5948<scan>parent layer:1, rssi:-17, assoc:1(cnx rssi threshold:-120)
I (42823) mesh: [SCAN][ch:11]AP:1, other(ID:0, RD:0), MAP:1, idle:0, candidate:1, root:1, topMAP:0[c:2,i:2][74:da:88:95:77:50][FIXED-ROOT]<weak>  
I (42833) mesh: 7370[weak]try rssi_threshold:-120, backoff times:0, max:5<-78,-82,-85>
I (42833) mesh: 716[monitor]no change, parent:34:85:18:8f:55:55, rssi:-17I (42843) mesh: 2004<arm>parent monitor, my layer:2(cap:6)(node), interval:313090ms, retries:3<>
I (43773) app_main: Heap size -> 8088952
I (44773) app_main: Heap size -> 8088952
I (45773) app_main: Heap size -> 8088952
I (46773) app_main: Heap size -> 8088796
I (47773) app_main: Heap size -> 8088796
W (47853) srv_ping: From x.x.x.x icmp_seq: 1 timeout
I (48773) app_main: Heap size -> 8088612
I (49773) app_main: Heap size -> 8088204
I (50773) app_main: Heap size -> 8088204
I (51773) app_main: Heap size -> 8088204
I (52773) app_main: Heap size -> 8088204
W (52853) srv_ping: From x.x.x.x icmp_seq: 2 timeout
I (53773) app_main: Heap size -> 8088204
I (54773) app_main: Heap size -> 8088000
I (55773) app_main: Heap size -> 8088000
I (56773) app_main: Heap size -> 8088000
I (57773) app_main: Heap size -> 8088000
W (57853) srv_ping: From x.x.x.x icmp_seq: 3 timeout
I (58773) app_main: Heap size -> 8087608
I (59773) app_main: Heap size -> 8087608
I (60773) app_main: Heap size -> 8087608
I (61773) app_main: Heap size -> 8087608
I (62773) app_main: Heap size -> 8087608
W (62853) srv_ping: From x.x.x.x icmp_seq: 4 timeout
```

atanisoft, 2024-09-10:
Do you have `CONFIG_LWIP_MULTICAST_PING` or `CONFIG_LWIP_BROADCAST_PING` enabled in your sdkconfig? If not, the nodes may not respond to a ping message.

atanisoft, 2024-09-10:
Any chance you can share a reproduction test case in code?

yanmartins, 2024-09-10:
@atanisoft I cant share the code, but in short, the implementation is a mix of these two examples:

- [IP Internal Network](https://github.com/espressif/esp-idf/tree/master/examples/mesh/ip_internal_network)
- [ICMP Echo](https://github.com/espressif/esp-idf/tree/master/examples/protocols/icmp_echo)

yanmartins, 2024-09-10:
My WiFi Mesh init
```c
bool srv_wifi_mesh_init(esp_netif_t            **netif_object,
                        bool                     wifi_sta_en,
                        wifi_sta_credentials_st *wifi_params,
                        wifi_mesh_config_st     *wifi_mesh_config,
                        uint8_t                  interface_id,
                        wifi_callback            callback,
                        mesh_rx_callback         rcv_callback)
{
    ESP_LOGI(TAG, "%s", __func__);

    wifi_cb = callback;

    if(wifi_mesh_initialized) return true;

    interface_id_sta = interface_id;

    if(!wifi_initialized) wifi_init(callback);

    //! Create a network interfaces for mesh (only station instance saved for further manipulation, soft AP instance ignored
    drv_mesh_netif_init(rcv_callback);

    esp_event_handler_register(IP_EVENT, IP_EVENT_STA_GOT_IP, &wifi_sta_ip_event_handler, NULL);

    esp_wifi_start();

    //! Mesh initialization
    esp_mesh_init();
    esp_mesh_set_xon_qsize(MESH_RX_WINDOW_SIZE);
    esp_mesh_allow_root_conflicts(MESH_ALLOW_ROOT_CONFLICTS);
    esp_mesh_send_block_time(MESH_SEND_BLOCK_TIME_MS);

    esp_mesh_fix_root(true);
    if(wifi_mesh_config->is_root) esp_mesh_set_type(MESH_ROOT);

    esp_event_handler_register(MESH_EVENT, ESP_EVENT_ANY_ID, &mesh_event_handler, NULL);
    esp_mesh_set_max_layer(MESH_MAX_LAYER);

    mesh_cfg_t cfg = MESH_INIT_CONFIG_DEFAULT();

    //! Mesh ID
    memcpy((uint8_t *) &cfg.mesh_id, wifi_mesh_config->id, WIFI_MESH_ID_LEN);

    //! Router
    if(wifi_sta_en)
    {
        wifi_sta_initialized = true;
        cfg.channel          = MESH_CHANNEL;
        cfg.router.ssid_len  = strlen((char *) wifi_params->ssid);
        memcpy((uint8_t *) &cfg.router.ssid, wifi_params->ssid, cfg.router.ssid_len);
        memcpy((uint8_t *) &cfg.router.password, wifi_params->password, sizeof(wifi_params->password));
    }

    //! Mesh softAP
    esp_mesh_set_ap_authmode(WIFI_AUTH_WPA2_PSK);
    cfg.mesh_ap.max_connection         = MESH_AP_MAX_CONNECTION;
    cfg.mesh_ap.nonmesh_max_connection = MESH_AP_MAX_NONMESH_CONNECTION;
    memcpy((uint8_t *) &cfg.mesh_ap.password, wifi_mesh_config->password, sizeof(wifi_mesh_config->password));
    esp_mesh_set_config(&cfg);

    print_wifi_mesh_config(&cfg);

    //! Mesh start
    esp_mesh_start();
    ESP_LOGI(TAG, "mesh starts successfully, heap: %lu, %s\n", esp_get_free_heap_size(), esp_mesh_is_root_fixed() ? "root fixed" : "root not fixed");
    wifi_mesh_initialized = true;

    return true;
}
```

My ping init:
```c
bool srv_ping_init(srv_ping_config_st *ping_configuration, ping_callback callback)
{
	ESP_LOGI(TAG, "%s", __func__);
    config = *ping_configuration;
	ping_cb = callback;
	esp_ping_config_t ping_config = (esp_ping_config_t) ESP_PING_DEFAULT_CONFIG();
	ping_config.task_stack_size = 1024 * 5;
	success_total_time = 0;

	//! Convert URL to IP address
	ip_addr_t target_addr;
	memset(&target_addr, 0, sizeof(target_addr));
	struct addrinfo hint;
	memset(&hint, 0, sizeof(hint));
	struct addrinfo *res = NULL;
	int err = getaddrinfo(config.target_host, NULL, &hint, &res);
	if(err != 0 || res == NULL)
	{
		ESP_LOGE(TAG, "DNS lookup failed err=%d res=%p", err, res);
		return false;
	} 
	else ESP_LOGI(TAG, "DNS lookup success");

	if (res->ai_family == AF_INET)
	{
		struct in_addr addr4 = ((struct sockaddr_in *) (res->ai_addr))->sin_addr;
		inet_addr_to_ip4addr(ip_2_ip4(&target_addr), &addr4);
	}
	else
	{
		struct in6_addr addr6 = ((struct sockaddr_in6 *) (res->ai_addr))->sin6_addr;
		inet6_addr_to_ip6addr(ip_2_ip6(&target_addr), &addr6);
	}
	freeaddrinfo(res);
	ESP_LOGI(TAG, "target_addr.type=%d", target_addr.type);
	ESP_LOGI(TAG, "target_addr.u_addr.ip4=%s", ip4addr_ntoa(&(target_addr.u_addr.ip4)));

	ping_config.target_addr = target_addr;
	ping_config.count = config.amount_of_pings;
	ping_config.interval_ms = config.interval_ms;
	ping_config.timeout_ms = config.timeout_ms;

	esp_ping_handle_t handle;

	esp_ping_callbacks_t cbs = {
		.cb_args = NULL,
		.on_ping_success = cmd_ping_on_ping_success,
		.on_ping_timeout = cmd_ping_on_ping_timeout,
		.on_ping_end = cmd_ping_on_ping_end
	};

    return (esp_ping_new_session(&ping_config, &cbs, &handle) == ESP_OK) && (esp_ping_start(handle) == ESP_OK);
}
```

yanmartins, 2024-09-17:
@zhangyanjiaoesp Do you have any suggestions about this?

zhangyanjiaoesp, 2024-09-18:
@yanmartins 
I apologize for not having had the time to review your code yet. However, the modified code provided by the client in [this link](https://github.com/espressif/esp-idf/issues/13212#issue-2141882899) can ping successfully. You can take a look. And I will review this issue ASAP.

yanmartins, 2024-09-18:
> @yanmartins I apologize for not having had the time to review your code yet. However, the modified code provided by the client in [this link](https://github.com/espressif/esp-idf/issues/13212#issue-2141882899) can ping successfully. You can take a look. And I will review this issue ASAP.

@zhangyanjiaoesp I have tried to make the modifications you made in the examples, but have not yet been successful

yanmartins, 2024-09-18:
One thing I don't know if I made clear is: the nodes don't lose connection, they are publishing data on MQTT, but I can't get them to ping. I want to do this to measure their latency

yanmartins, 2024-10-14:
Hello @zhangyanjiaoesp. Do you have any suggestions about this?

zhangyanjiaoesp, 2025-01-13:
@yanmartins 
Please provide a demo for me to test, based on the available information, I am unable to determine the cause of the ping failure.

Sherry616, 2025-08-04:
Hi @yanmartins, do you have any updates on this issue? Thanks!