espressif/esp-idf #14539: generate pwm signal with sub hz frequency

a-3isa, 2024-09-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.0

### Espressif SoC revision.

esp-32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp-32

### Power Supply used.

USB

### What is the expected behavior?

a pwm of 0.1 hz frequency

### What is the actual behavior?

the program hangs and is not working

### Steps to reproduce.

use the ledc_basic example and make the frequency = 0.1 hz

songruo, 2024-09-10:
@a-3isa In our current LEDC driver, `ledc_timer_config_t :: freq_hz` is with type of `uint32_t`, it cannot generate a frequency less than 1 Hz. Maybe you can try with the RMT driver, it can be used to generate a 0.1Hz PWM.

a-3isa, 2024-09-14:
Will it work if i modify the type to double?

a-3isa, 2024-09-14:
Will it work if i modify the type to `double` ?