espressif/esp-idf #14542: unit_test repository doesn't run tests once "testable" components has been renamed

broccan, 2024-09-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1-1-gcc3203dc4f

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Running `idf.py -p COM1 flash monitor` should produce the following no matter what the component name is.
```
#### Executing one test by its name #####

Running Mean of an empty array is zero...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:16:Mean of an empty array is zero:PASS

-----------------------
1 Tests 0 Failures 0 Ignored
OK

#### Running tests with [mean] tag #####

Running tests matching '[mean]'...
Running Mean of an empty array is zero...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:16:Mean of an empty array is zero:PASS
Running Mean of a test vector...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:22:Mean of a test vector:PASS
Running Another test case which fails...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:35:Another test case which fails:FAIL: Expected 2147483647 Was -1. Function [mean][fails]

-----------------------
3 Tests 1 Failures 0 Ignored
FAIL

#### Running tests without [fails] tag #####

Running tests NOT matching '[fails]'...
Running Mean of an empty array is zero...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:16:Mean of an empty array is zero:PASS
Running Mean of a test vector...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:22:Mean of a test vector:PASS

-----------------------
2 Tests 0 Failures 0 Ignored
OK

#### Running all the registered tests #####

Running Mean of an empty array is zero...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:16:Mean of an empty array is zero:PASS
Running Mean of a test vector...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:22:Mean of a test vector:PASS
Running Another test case which fails...
/IDF/examples/system/unit_test/components/testable2/test/test_mean.c:35:Another test case which fails:FAIL: Expected 2147483647 Was -1. Function [mean][fails]

-----------------------
3 Tests 1 Failures 0 Ignored
FAIL

#### Starting interactive test menu #####

Press ENTER to see the list of tests.
```

### What is the actual behavior?

What actually happens is the tests don't run unless the component being tested is named "testable", so if I change the component name to "testable2" I get the following output when running "idf.py -p COM1 flash monitor".
```
#### Executing one test by its name #####

-----------------------
0 Tests 0 Failures 0 Ignored
OK

#### Running tests with [mean] tag #####

Running tests matching '[mean]'...

-----------------------
0 Tests 0 Failures 0 Ignored
OK

#### Running tests without [fails] tag #####

Running tests NOT matching '[fails]'...

-----------------------
0 Tests 0 Failures 0 Ignored
OK

#### Running all the registered tests #####

-----------------------
0 Tests 0 Failures 0 Ignored
OK

#### Starting interactive test menu #####

Press ENTER to see the list of tests.
```

### Steps to reproduce.

1. Install esp-idf
2. cd to examples/system/unit_test
3. rename "testable" component to "testable2"
4. modify "test\CMakeLists.txt" file to:
```
 This is the project CMakeLists.txt file for the test subproject
cmake_minimum_required(VERSION 3.16)

# Include the components directory of the main application:
#
set(EXTRA_COMPONENT_DIRS "../components")

# Set the components to include the tests for.
# This can be overriden from CMake cache:
# - when invoking CMake directly: cmake -D TEST_COMPONENTS="xxxxx" ..
# - when using idf.py: idf.py -T xxxxx build
#
set(TEST_COMPONENTS "testable2" CACHE STRING "List of components to test")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(unit_test_test)
```
5. modify "components\testable\test\CMakeLists.txt"  to:
```
idf_component_register(SRC_DIRS "."
                    INCLUDE_DIRS "."
                    REQUIRES cmock testable2)
```
6. in test directory run `idf.py -p COM1 flash monitor`

### More Information.

Based on the solution: https://stackoverflow.com/questions/78678181/esp-idf-unit-tests-with-unity-and-c

If you add the line `set(EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/unit-test-app/components" "main")` to "unit_test/CMakeLists.txt" then clean and recompile it works. i.e.

1. Modify "unit_test/CMakeLists.txt" to:
```
# This is the project CMakeLists.txt file for the test subproject
cmake_minimum_required(VERSION 3.16)

set(EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/unit-test-app/components" "main")

include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(unit_test)
```
2. in "test" directory run: `idf.py fullclean`
3. in "test" directory run: `idf.py -p COM1 flash monitor`

Then the unit_test example works for component names other than "testable"

KonstantinKondrashov, 2025-04-18:
Hi @broccan!
Sorry, we missed this issue. Seems you already provided the solution for it.
If you rename a component, it seems reasonable that you have to update the name in the CMake files and do a full rebuild.

Rename a component -> update Cmake files -> `idf.py fullclean` -> `idf.py flash monitor`

```diff
diff --git a/examples/system/unit_test/components/testable/test/CMakeLists.txt b/examples/system/unit_test/components/testable/test/CMakeLists.txt
index f5b82bc143c..2d266364681 100644
--- a/examples/system/unit_test/components/testable/test/CMakeLists.txt
+++ b/examples/system/unit_test/components/testable/test/CMakeLists.txt
@@ -1,3 +1,3 @@
 idf_component_register(SRC_DIRS "."
                     INCLUDE_DIRS "."
-                    REQUIRES cmock testable)
+                    REQUIRES cmock testable2)
diff --git a/examples/system/unit_test/test/CMakeLists.txt b/examples/system/unit_test/test/CMakeLists.txt
index 77c280b2e7c..0adc076ffac 100644
--- a/examples/system/unit_test/test/CMakeLists.txt
+++ b/examples/system/unit_test/test/CMakeLists.txt
@@ -10,7 +10,7 @@ set(EXTRA_COMPONENT_DIRS "../components")
 # - when invoking CMake directly: cmake -D TEST_COMPONENTS="xxxxx" ..
 # - when using idf.py: idf.py -T xxxxx build
 #
-set(TEST_COMPONENTS "testable" CACHE STRING "List of components to test")
+set(TEST_COMPONENTS "testable2" CACHE STRING "List of components to test")
 
 include($ENV{IDF_PATH}/tools/cmake/project.cmake)
 project(unit_test_test)
```

closed 2025-04-18 by KonstantinKondrashov

broccan, 2025-04-18:
Hello @KonstantinKondrashov,

The reason for this issue is that the example unity test reposity given in this repository only builds if the component name is testable. In extra information I have given a line that needs to be added to a CMakeLists.txt file to get it to build. The full clean and rebuilding steps are of course required and for clarity I have included them, but this extra line is the purpose of the issue. I hope that an edit of the unity example repository will save future users a lot of time working with unity tests and esp-idf for the first time.

Note: I have not tested the issue again with recent versions of esp-idf so it may no longer require modification for the example project to work.

Thank you for your work