espressif/esp-idf #14543: nimble-mesh LoadProhibited

Laosepy, 2024-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release v5.3.1

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32 Devkit V1

### Power Supply used.

USB

### What is the expected behavior?

I used the mesh routine in the nimble example and I wanted it to be scanned and provision
[image]

### What is the actual behavior?

In fact, I used the nrf connect app to scan the device after compiling and writing. I found that it did not send any broadcast at all. After consulting the source code (), I found that bt_mesh_init needs to be enabled by calling bt_mesh_prov_enable. But I got a stack error when I added this function

### Steps to reproduce.

1. use nimble mesh example code
2. add bt_mesh_prov_enable after bt_mesh_init with blemesh_on_sync

static void
blemesh_on_sync(void)
{
    int err;
    ble_addr_t addr;

    ESP_LOGI(tag, "Bluetooth initialized");

    /* Use NRPA */
    err = ble_hs_id_gen_rnd(1, &addr);
    assert(err == 0);
    err = ble_hs_id_set_rnd(addr.val);
    assert(err == 0);

    err = bt_mesh_init(addr.type, &prov, &comp);
    if (err) {
        ESP_LOGI(tag, "Initializing mesh failed (err %d)", err);
        return;
    }

#if (MYNEWT_VAL(BLE_MESH_SHELL))
    shell_register_default_module("mesh");
#endif

    ESP_LOGI(tag, "Mesh initialized");

    if (IS_ENABLED(CONFIG_SETTINGS)) {
        settings_load();
    }

    if (bt_mesh_is_provisioned()) {
        ESP_LOGI(tag, "Mesh network restored from flash");
    }

    bt_mesh_prov_enable(BT_MESH_PROV_GATT | BT_MESH_PROV_ADV);
}

### Debug Logs.

```plain
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM21 instead...
--- esp-idf-monitor 1.4.0 on \\.\COM21 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�L      �[0m
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (31) boot: compile time Sep 11 2024 10:26:04
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1cea4h (118436) map
I (144) esp_image: segment 1: paddr=0002cecc vaddr=3ffbdb60 size=0314ch ( 12620) load
I (149) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=621b0h (401840) map
I (288) esp_image: segment 3: paddr=000921d8 vaddr=3ffc0cac size=01f50h (  8016) load
I (291) esp_image: segment 4: paddr=00094130 vaddr=40080000 size=175c8h ( 95688) load
I (343) boot: Loaded app from partition at offset 0x10000
I (343) boot: Disabling RNG early entropy source...
I (355) cpu_start: Multicore app
I (363) cpu_start: Pro cpu start user code
I (363) cpu_start: cpu freq: 160000000 Hz
I (364) app_init: Application information:
I (366) app_init: Project name:     blemesh
I (371) app_init: App version:      1
I (376) app_init: Compile time:     Sep 11 2024 10:25:34
I (382) app_init: ELF file SHA256:  64751b274...
I (387) app_init: ESP-IDF:          v5.3.1
I (392) efuse_init: Min chip rev:     v0.0
I (396) efuse_init: Max chip rev:     v3.99
I (401) efuse_init: Chip rev:         v1.0
I (407) heap_init: Initializing. RAM available for dynamic allocation:
I (414) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (419) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (426) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (432) heap_init: At 3FFC7B68 len 00018498 (97 KiB): DRAM
I (438) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (444) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (451) heap_init: At 400975C8 len 00008A38 (34 KiB): IRAM
I (459) spi_flash: detected chip: generic
I (461) spi_flash: flash io: dio
I (466) coexist: coex firmware version: 4482466
I (471) main_task: Started on CPU0
I (481) main_task: Calling app_main()
I (481) BTDM_INIT: BT controller compile version [b022216]
I (491) BTDM_INIT: Bluetooth MAC: 84:cc:a8:7a:ee:86
I (491) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
I (831) NimBLE_MESH: Bluetooth initialized
I (911) NimBLE_MESH: Mesh initialized
I (911) NimBLE: GAP procedure initiated: discovery;
I (911) NimBLE: own_addr_type=1 filter_policy=0 passive=1 limited=0 filter_duplicates=0
I (921) NimBLE: duration=forever
I (921) NimBLE: 

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e14f1  PS      : 0x00060330  A0      : 0x800f206c  A1      : 0x3ffd0f10  
0x400e14f1: npl_freertos_eventq_put at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:304

A2      : 0x3ffc4634  A3      : 0x3ffc6a60  A4      : 0x3ffd0f40  A5      : 0x00000000
A6      : 0x3f4036cc  A7      : 0x00000001  A8      : 0x00000000  A9      : 0x3ffd0ed0
A10     : 0x00000008  A11     : 0x00000019  A12     : 0x3f40414c  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000007  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
0x4000c46c: memset in ROM
0x4000c477: memset in ROM

Backtrace: 0x400e14ee:0x3ffd0f10 0x400f2069:0x3ffd0f50 0x400f5d12:0x3ffd0f70 0x400e7522:0x3ffd0f90 0x400d83aa:0x3ffd0fb0 0x400d9173:0x3ffd0fe0 0x400d931b:0x3ffd1000 0x400d9323:0x3ffd1020 0x4012d6f3:0x3ffd1040 0x40083382:0x3ffd1060 0x400d8452:0x3ffd1080 0x4009003d:0x3ffd10a0
0x400e14ee: npl_freertos_eventq_put at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:301
0x400f2069: ble_npl_eventq_put at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:173
 (inlined by) bt_mesh_adv_update at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/nimble/host/mesh/src/adv_legacy.c:195
0x400f5d12: link_accept at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/nimble/host/mesh/src/pb_gatt.c:131
0x400e7522: bt_mesh_prov_enable at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/nimble/host/mesh/src/prov_device.c:605 (discriminator 1)
0x400d83aa: blemesh_on_sync at E:/Espressif/BLE_mesh/nimble_mesh/blemesh/main/app_mesh.c:432
0x400d9173: ble_hs_sync at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:373
0x400d931b: ble_hs_start at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:681
0x400d9323: ble_hs_event_start_stage2 at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/nimble/host/src/ble_hs.c:581
0x4012d6f3: npl_freertos_event_run at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/porting/npl/freertos/src/npl_os_freertos.c:451
0x40083382: ble_npl_event_run at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/porting/npl/freertos/include/nimble/nimble_npl_os.h:185
 (inlined by) nimble_port_run at D:/Espressif/frameworks/esp-idf-v5.3.1/components/bt/host/nimble/nimble/porting/nimble/src/nimble_port.c:306
0x400d8452: blemesh_host_task at E:/Espressif/BLE_mesh/nimble_mesh/blemesh/main/app_mesh.c:442
0x4009003d: vPortTaskWrapper at D:/Espressif/frameworks/esp-idf-v5.3.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 64751b274
```

### More Information.

non

rahult-github, 2024-09-11:
Hi @Laosepy ,

As mentioned in documentation,  the example is a copy from upstream mynewt-nimble and is "Not maintained by  Espressif" .     

If you want to use Ble Mesh, request you to please use [ESP-BLE-MESH ](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh) which is officially supported and maintained by Espressif.

Laosepy, 2024-09-11:
i want to use ble mesh and wifi sta coex with esp32-c2 ,but None of the sample code works properly, is bug, or should be removed from the sample

rahult-github, 2024-09-11:
> i want to use ble mesh and wifi sta coex with esp32-c2 ,but None of the sample code works properly, is bug, or should be removed from the sample
 
 For this, are you using the esp-ble-mesh example ?

Laosepy, 2024-09-11:
Sorry, my description is not clear enough, my ultimate goal is to want to use ble mesh and wifi sta coex with esp32-c2, However, esp sales engineer said that esp32-c2 cannot run ESP-ble-mesh and wifi sta coex, only nimble mesh and wifi sta coex, so I am debugging for nimble mesh now

Laosepy, 2024-09-12:
> Sorry, my description is not clear enough, my ultimate goal is to want to use ble mesh and wifi sta coex with esp32-c2, However, esp sales engineer said that esp32-c2 cannot run ESP-ble-mesh and wifi sta coex, only nimble mesh and wifi sta coex, so I am debugging for nimble mesh now

Laosepy, 2024-09-12:
> > i want to use ble mesh and wifi sta coex with esp32-c2 ,but None of the sample code works properly, is bug, or should be removed from the sample
> 
> For this, are you using the esp-ble-mesh example ?

Sorry, my description is not clear enough, my ultimate goal is to want to use ble mesh and wifi sta coex with esp32-c2, However, esp sales engineer said that esp32-c2 cannot run ESP-ble-mesh and wifi sta coex, only nimble mesh and wifi sta coex, so I am debugging for nimble mesh now

chegewara, 2024-09-30:
C2 cant run ble mesh bluedroid stack and wifi because C2 has very few of RAM, and to be honest it may be very hard to run wifi and ble mesh with nimble stack too.
Im not sure if i would try to do it on C2 myself.

buganini, 2024-10-05:
@Laosepy 

I got nimble mesh adv working by

using this patch in `esp-idf/components/bt/host/nimble/nimble`
``` diff
diff --git a/nimble/host/mesh/src/glue.c b/nimble/host/mesh/src/glue.c
index 5bce3d8d..1d0a534d 100644
--- a/nimble/host/mesh/src/glue.c
+++ b/nimble/host/mesh/src/glue.c
@@ -81,7 +81,7 @@ net_buf_put(struct ble_npl_eventq *fifo, struct os_mbuf *om)
     assert(OS_MBUF_IS_PKTHDR(om));
     ev = &BT_MESH_ADV(om)->ev;
     assert(ev);
-    assert(ble_npl_event_get_arg(ev));
+    // assert(ble_npl_event_get_arg(ev));

     ble_npl_eventq_put(fifo, ev);
 }
diff --git a/porting/npl/freertos/src/npl_os_freertos.c b/porting/npl/freertos/src/npl_os_freertos.c
index d72f0f6e..ee150ead 100644
--- a/porting/npl/freertos/src/npl_os_freertos.c
+++ b/porting/npl/freertos/src/npl_os_freertos.c
@@ -301,11 +301,11 @@ npl_freertos_eventq_put(struct ble_npl_eventq *evq, struct ble_npl_event *ev)
     struct ble_npl_eventq_freertos *eventq = (struct ble_npl_eventq_freertos *)evq->eventq;
     struct ble_npl_event_freertos *event = (struct ble_npl_event_freertos *)ev->event;

-    if (event->queued) {
-        return;
-    }
+    // if (event->queued) {
+    //     return;
+    // }

-    event->queued = true;
+    // event->queued = true;

     if (in_isr()) {
         ret = xQueueSendToBackFromISR(eventq->q, &ev, &woken);
```

and 
``` C
    void mesh_adv_thread(void *args);
    mesh_adv_thread(NULL);
```
at the end of your main function, or start a task for this.

`mesh_adv_thread` is defined in `esp-idf/components/bt/host/nimble/nimble/nimble/host/mesh/src/adv_legacy.c`
and should be started (but not) by
``` C
#ifdef MYNEWT
	os_task_init(&adv_task, "mesh_adv", mesh_adv_thread, NULL,
	             MYNEWT_VAL(BLE_MESH_ADV_TASK_PRIO), OS_WAIT_FOREVER,
	             g_blemesh_stack, MYNEWT_VAL(BLE_MESH_ADV_STACK_SIZE));
#endif
```

rahult-github, 2024-12-18:
Hi @Laosepy  , as mentioned in readme, this example is copy of the code from upstream mynewt-nimble and is not maintained by Espressif.  Instead, we recommend to use [ESP-BLE-MESH](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/esp_ble_mesh) solution , which is espressif's official mesh solution. Request you to please use the same.

rahult-github, 2024-12-30:
Closing this issue. Feel free to raise a new issue if there are any issues/concerns on ESP-BLE-MESH solution usage

closed 2024-12-30 by rahult-github