espressif/esp-idf #14545: OTA chunk size problems

jkorenj, 2024-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, 

I implemented OTA update through SPI on ESP32C3. 
As I read in the documentation the flash page size is 4k, but for some reason when I try to send 4k chunks the partition is never changed. 
I can see that the update procedure is successful, but when I reset the ESP32C3 it is booted from the same partition. 
I tried it with 512 bytes chunk size and everything works as expected. 

Can you help me, what am I missing here? 

Thank you for your help.

nileshkale123, 2024-10-07:
Hello @jkorenj ,

I'm sorry for the delayed reply. Thank you for your patience.

Could you please let me know which example you have tried for implementing the OTA update on the ESP32-C3? Additionally, it would be helpful if you could share the sdkconfig file and the debug-enabled log (CONFIG_LOG_DEFAULT_LEVEL_DEBUG) to get more insight into the issue.

This will help us understand the situation better and provide a more accurate solution.

Looking forward to your response.

jkorenj, 2024-10-10:
Hello, 

I implemented OTA update through SPI and I didn't use any example, but I was looking at the simple_ota_example. 
Problem is that everything seems normal and status from update function is returned as everything was correctly run, but ESP doesn't reset and partition doesn't change. 

I can prepare the sample in the next week. 

Thank you.

nileshkale123, 2024-10-15:
Hello @jkorenj,

Alternatively, you can integrate your changes into the simple_ota_example and try to reproduce the issue. If the problem persists, please share the patch and sdkconfig file with us so we can reproduce it on our end.

Thank you.

nileshkale123, 2024-10-22:
Here’s a more polished version of the message:

Hi @jkorenj,

Since we haven't received any further updates, I’ll be closing this issue for now.

However, please feel free to reopen it if you encounter the issue again. It would be helpful if you could provide a small reproducible example or a patch file integrated with one of the OTA examples in IDF.

Best regards,