espressif/esp-idf #14546: [v4.4] [mesh] Infinite loop MESH_EVENT_ROOT_FIXED

KonssnoK, 2024-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hello @zhangyanjiaoesp 
somehow i triggered the following infinite loop on one test device
[240911_infiniteloop.txt](https://github.com/user-attachments/files/16963672/240911_infiniteloop.txt)

the device constantly receives the MESH_EVENT_ROOT_FIXED event with the fixed flag toggling continuously.

I have no idea how this was triggered..

zhangyanjiaoesp, 2024-09-12:
@KonssnoK 
Two scenarios can trigger the `MESH_EVENT_ROOT_FIXED` event at the mesh layer: 

1. A change is detected in the parent’s IE
2. The `esp_mesh_fix_root()` API is called

From your log, I am unable to determine why the root_fixed flag of the event is consistently changing.

KonssnoK, 2024-09-12:
so what happened was that i was testing  mesh with 4 devices where one was periodically resetting the mesh library. This device was actually not doing anything special but scanning for neighbors and having to follow the other ones..

The other 3 devices were behaving correctly and all connected when this was in the infinite loop, so we can exclude both 1 and 2...

zhangyanjiaoesp, 2025-02-13:
@KonssnoK Do you still have this problem? Can we close it?

Alvin1Zhang, 2025-02-25:
Thanks for reporting, will close due to inactivity, feel free to reopen if the issue still happens, thanks.

closed 2025-02-25 by Alvin1Zhang