espressif/esp-idf #14549: esp_mesh_send 发送失败，如何清除包内容？

Leadrive, 2024-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.6

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

调用 esp_mesh_send 后，能清除发送内容，而不是失败后累积起来

### What is the actual behavior?

````
static esp_err_t mesh_netif_transmit_from_node_sta(void *h, void *buffer, size_t len)
{
    if (len>=CONFIG_LWIP_TCP_MSS) {
        MDF_LOGE("Sending to root, dest addr: " MACSTR ", size: %d" ,MAC2STR((uint8_t*)buffer), len);
        return ESP_ERR_MESH_ARGUMENT;
    }
    mesh_data_t data;
    data.data = buffer;
    data.size = len;
    data.proto = MESH_PROTO_AP; // Node's station transmits data to root's AP
//    data.tos = MESH_TOS_DEF;
    data.tos = MESH_TOS_P2P;
    esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS, NULL, 0);
    if (err != ESP_OK) {
        MDF_LOGI("Sending to root, dest addr: " MACSTR ", size: %d" ,MAC2STR((uint8_t*)buffer), len);
        MDF_LOGW("%s, err=<%s>, 内部堆=%u, 外部堆=%u", __func__, mdf_err_to_name(err), esp_get_free_internal_heap_size(), esp_get_free_heap_size());
        if (err==ESP_ERR_MESH_ARGUMENT) {
            MDF_LOGI("%.*s", len, (char*)buffer);
        }
    }
    return err;
}
````
此时，如果连不上ROOT，发送失败，发送的内容，并没有清除，而是累积起来，最终导致长度超过1440而报错。
如果参数改成
data.tos = MESH_TOS_DEF;
则会时不时报
ESP_ERR_MESH_ARGUMENT

[... 121 lines cut ...]
E (1726062229) [mesh_netif, 223]: Sending to root, dest addr: e0:5a:1b:15:0a:39, size: 1478
E (1726062229) [mesh_netif, 223]: Sending to root, dest addr: e0:5a:1b:15:0a:39, size: 1478
E (1726062230) [mesh_netif, 223]: Sending to root, dest addr: e0:5a:1b:15:0a:39, size: 1478
I (1244248) wifi:state: auth -> init (200)
I (1244249) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
I (1244251) mesh: [wifi]disconnected reason:2(auth expire), continuous:0/max:12, non-root, vote(,)<><>
I (1726062230) [prj_init, 817]: SetNetworkConnected, DirectNetConnected=0
I (1244261) mesh: [auth expire]looking for a new parent, [L:2]try layer:1
I (1726062230) [mesh_base, 301]: Network state: root_less
E (1726062230) [mesh_netif, 223]: Sending to root, dest addr: e0:5a:1b:15:0a:39, size: 1478
I (1244527) mesh: [SCAN][ch:1]AP:8, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (1244529) mesh: [FAIL][1]root:0, fail:1, normal:0

I (1244776) mesh: [SCAN][ch:1]AP:7, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (1244777) mesh: [FAIL][2]root:0, fail:2, normal:0

E (1726062230) [mesh_netif, 223]: Sending to root, dest addr: e0:5a:1b:15:0a:39, size: 1478
I (1245024) mesh: [SCAN][ch:1]AP:7, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (1245026) mesh: [FAIL][3]root:0, fail:3, normal:0

E (1726062231) [mesh_netif, 223]: Sending to root, dest addr: e0:5a:1b:15:0a:39, size: 1478
I (1245273) mesh: [SCAN][ch:1]AP:8, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][00:00:00:00:00:00][FIXED-ROOT]<>
I (1245275) mesh: [FAIL][4]root:0, fail:4, normal:0
````
```

zhangyanjiaoesp, 2024-10-22:
@Leadrive 
For the configuration of the `esp_mesh_send()` function parameters, please read the document carefully.
https://github.com/espressif/esp-idf/blob/a97a7b0962da148669bb333ff1f30bf272946ade/components/esp_wifi/include/esp_mesh.h#L652

For the other issue, I will check it .

zhangyanjiaoesp, 2024-12-04:
> ### Steps to reproduce.
> 
> 用自带的 ip_internal_network 再修改一下，不断发一个什么 包内容。
> 连上ROOT后，一切运行正常。 
> 此时手工断开ROOT，观察NODE节点，就会发现esp_mesh_send发送的包，大小不断变大。

@Leadrive 我用`ip_internal_network` 测试，添加了你上面提到的 log，但是我没有复现你的问题。我的测试log如下：
请你提供一个可以复现问题的demo， 谢谢。
```
>>>Sending to root, dest addr: 34:85:18:a5:b1:81, size: 93

>>>mesh_netif_transmit_from_node_sta, 内部堆=208184, 外部堆=215688
I (34063) wifi:state: init -> auth (0xb0)
E (34063) mesh_netif: Send with err code 16395 ESP_ERR_MESH_DISCONNECTED
I (34063) mesh_mqtt: sent publish returned msg_id=38955
>>>Sending to root, dest addr: 34:85:18:a5:b1:81, size: 93
>>>mesh_netif_transmit_from_node_sta, 内部堆=207948, 外部堆=215452
E (34093) mesh_netif: Send with err code 16395 ESP_ERR_MESH_DISCONNECTED
>>>Sending to root, dest addr: 34:85:18:a5:b1:81, size: 350
>>>mesh_netif_transmit_from_node_sta, 内部堆=207724, 外部堆=215228
E (34113) mesh_netif: Send with err code 16395 ESP_ERR_MESH_DISCONNECTED
>>>Sending to root, dest addr: 34:85:18:a5:b1:81, size: 54
>>>mesh_netif_transmit_from_node_sta, 内部堆=208772, 外部堆=216276
E (34133) transport_base: poll_read select error 113, errno = Software caused connection abort, fd = 54
E (34143) mqtt_client: Poll read error: 119, aborting connection
I (34143) mesh_mqtt: MQTT_EVENT_DISCONNECTED
I (35283) wifi:state: auth -> init (0x200)

```

zhangyanjiaoesp, 2024-12-23:
@Leadrive 这个问题还存在吗？是否可以关闭？

closed 2025-01-07 by Alvin1Zhang