espressif/esp-idf #14552: wifi_prov_mgr_reset_sm_state_on_failure() does not restart provisioning

kparshall-altus, 2024-09-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

When running the `wifi_prov_mgr` demo app with `CONFIG_EXAMPLE_RESET_PROV_MGR_ON_FAILURE` enabled in menuconfig, incorrectly entering a station's credentials will call `wifi_prov_mgr_reset_sm_state_on_failure()` on line 144 which restarts provisioning without requiring a restart of the device. 

### What is the actual behavior?

While the function runs and returns an ESP_OK, the state machine does not recover and the device ID cannot be found in the  ESP BLE Prov app.

### Steps to reproduce.

1. Run the wifi_prov_mgr demo app on an esp32-s3. Security is set to version 1.
2. Connect to the device using the esp ble prov demo app using ble
3. Enter PoP and scan for wifi networks, but enter an incorrect password for a given SSID.
4. App will show an error. Hit okay and then go back to the bluetooth menu.
5. Device does not show up. 

### Debug Logs.

```plain
x buffer size: 1600
I (538) wifi:Init static rx buffer num: 10
I (538) wifi:Init dynamicESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d98
entry 0x403c9914
I (27) boot: ESP-IDF v5.2-dirty 2nd stage bootloader
I (27) boot: compile time Aug 28 2024 15:50:10
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
[... 169 lines cut ...]
E (111708) app: Provisioning failed!
        Reason : Wi-Fi station authentication failed
        Please reset to factory and retry provisioning
I (115068) app: Disconnected. Connecting to the AP again...
E (115068) wifi_prov_mgr: STA Disconnected
E (115068) wifi_prov_mgr: Disconnect reason : 205
E (115078) wifi:sta is connecting, return error
I (118438) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
I (118448) wifi:state: init -> auth (b0)
I (118448) wifi:state: auth -> assoc (0)
I (118448) wifi:state: assoc -> run (10)
I (118468) wifi:<ba-add>idx:0 (ifx:0, f0:a7:31:c8:8b:eb), tid:5, ssn:0, winSize:64
I (121548) wifi:state: run -> init (fc0)
I (121558) wifi:<ba-del>idx:0, tid:5
I (121558) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1
I (121558) app: Disconnected. Connecting to the AP again...
E (121558) wifi_prov_mgr: STA Disconnected
E (121568) wifi_prov_mgr: Disconnect reason : 15
E (121568) wifi_prov_mgr: STA Auth Error
E (121578) app: Provisioning failed!
        Reason : Wi-Fi station authentication failed
        Please reset to factory and retry provisioning
I (121588) app: Failed to connect with provisioned AP, reseting provisioned credentials
I (124938) app: Disconnected. Connecting to the AP again...
```

hrushikesh430, 2024-09-17:
Hello @kparshall-altus, the network list will be shown again after number of max retries (max retries is set in the menuconfig). For looking how it works you can refer [this](https://github.com/espressif/esp-idf/blob/v5.3.1/examples/provisioning/wifi_prov_mgr/main/app_main.c#L136) example code.

kparshall-altus, 2024-09-27:
Hello, it looks like the system does not recover though. Line 144 of the example calls the function:
wifi_prov_mgr_reset_sm_state_on_failure();

When operating the demo app and entering incorrect credentials I have verified that the function is called.

It is my understanding that this function will reset the state machine which does not appear to occur. Do I need to call something to restart provisioning as well?

My hope is that someone who enters an incorrect password does not have to restart the device to retry.

hrushikesh430, 2024-09-30:
@kparshall-altus After attempting max retries, the wifi_prov_mgr_reset_sm_state_on_failure() resets the provisioning state to WIFI_PROV_STATE_STARTED and sets the wifi_configs again. So that you just need to scan the previously generated QR and do provisioning. Just check it by scanning older QR if it works from your side.

kparshall-altus, 2024-09-30:
Hrushikesh430, I should have clarified that I am trying to use bluetooth, not scanning the QR code. Does wifi_prov_mgr_reset_sm_state_on_failure() only work with the QR code?

hrushikesh430, 2024-09-30:
you do provisioning using BLE right?

hrushikesh430, 2024-09-30:
@kparshall-altus no it is irrespective of whether you use QR or have direct connection using bluetooth. I have confirmed it from my side it works fine. When you put incorrect password and click ok on your app and go back. Try to connect your device using Bluetooth again and do provisioning.

hrushikesh430, 2024-10-14:
@kparshall-altus I am closing this issue due to no response. If any quires or issue you get, feel free to reopen this issue.

closed 2024-10-14 by hrushikesh430