espressif/esp-idf #14553: esp-idf toolsoffiline 4.3.2安装开发环境失败

Tracker647, 2024-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-4.3.2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

esp idf 4.3.2环境安装成功

### What is the actual behavior?

[image]

### Steps to reproduce.

1. Step
从https://github.com/espressif/idf-installer/releases?page=9获取4.3.2的offline installer
2. Step
安装下载，下载路径为D:/Devtools
3.Step
所有步骤完成后，在安装ftdi驱动部分报错
...

### Build or installation Logs.

```plain
看图片
```

### More Information.

项目版本原因，只能用v4.3.2，以前电脑能正常下载，最近换了新电脑后就发现不能下载了，下载最新的idf版本是正常的，在公司的电脑上尝试下v4.3.2也是报错，看build log的原因都和ftdi驱动有关，是我OS的问题吗？

jakub-kocka, 2024-09-12:
Hello, @Tracker647,

We are sorry for the inconvenience. You are right that the Installer for v4.3.2 is not working correctly, I was able to reproduce your issue. The issue is with the FTDI driver files bundled with the Installer itself.

### Solution 1
If you don't rely on the PATCH version that much, you can use [IDF v4.3.6 Installer](https://github.com/espressif/idf-installer/releases/download/offline-4.3.6/esp-idf-tools-setup-offline-4.3.6.exe) (which should your project be working on as well) the installation should be OK with this.

### Solution 2
You can use the [IDF v4.3.2 Installer](https://github.com/espressif/idf-installer/releases/download/offline-4.3.2/esp-idf-tools-setup-offline-4.3.2.exe) but a few more steps are needed:

* Please download the FTDI driver separately and install it
    - FTDI download: https://www.ftdichip.com/Driver/CDM/CDM%20v2.12.28%20WHQL%20Certified.zip
    - Extract the downloaded archive
    - Right-click the` ftdibus.inf` file and select **`Install`** option
    
* Then start the ESP-IDF installation with the ESP-IDF v4.3.2 Windows Installer
* During the installation process, uncheck the FTDI driver installation

[image]

And that should be it, the installation process should finish with success this time.

I hope this will help you.

Tracker647, 2024-09-12:
> Hello, @Tracker647,
> 
> We are sorry for the inconvenience. You are right that the Installer for v4.3.2 is not working correctly, I was able to reproduce your issue. The issue is with the FTDI driver files bundled with the Installer itself.
> 
> ### Solution 1
> If you don't rely on the PATCH version that much, you can use [IDF v4.3.6 Installer](https://github.com/espressif/idf-installer/releases/download/offline-4.3.6/esp-idf-tools-setup-offline-4.3.6.exe) (which should your project be working on as well) the installation should be OK with this.
> 
> ### Solution 2
> You can use the [IDF v4.3.2 Installer](https://github.com/espressif/idf-installer/releases/download/offline-4.3.2/esp-idf-tools-setup-offline-4.3.2.exe) but a few more steps are needed:
> 
> * Please download the FTDI driver separately and install it
>   
>   * FTDI download: https://www.ftdichip.com/Driver/CDM/CDM%20v2.12.28%20WHQL%20Certified.zip
>   * Extract the downloaded archive
>   * Right-click the` ftdibus.inf` file and select **`Install`** option
> * Then start the ESP-IDF installation with the ESP-IDF v4.3.2 Windows Installer
> * During the installation process, uncheck the FTDI driver installation
> 
> <img alt="image" width="449" src="https://private-user-images.githubusercontent.com/46049162/366740079-7624cd15-cdac-41bc-b4bf-87853328e9c4.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjYxMjM5NTMsIm5iZiI6MTcyNjEyMzY1MywicGF0aCI6Ii80NjA0OTE2Mi8zNjY3NDAwNzktNzYyNGNkMTUtY2RhYy00MWJjLWI0YmYtODc4NTMzMjhlOWM0LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA5MTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwOTEyVDA2NDczM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZhN2Q2MTIxNzg4NGM3NjBlNGNiYzY0ZDE5ODUzNzBiZGMzOWE3NzY3ZTYzOWEyYjY3NGFkOTllNTMxODZkODAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.MzgKs-Nti9cHjRuNnoBVI1tX-Lqdkl9MOu9I14hDSHI">
> And that should be it, the installation process should finish with success this time.
> 
> I hope this will help you.

it works, thanks.

closed 2024-09-12 by Tracker647