espressif/esp-idf #14554: How to upload nvs partition with wifi credentials

ConDai, 2024-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I would like to generate a nvs partition containing wifi credentials and upload it to my ESP32. 
I am generating the nvs partition from a CSV using the [NVS Partition Generator](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_partition_gen.html)

I want to set the wifi credentials in the CSV in a format compatible with the esp wifi component. 
The esp wifi saves wifi credentials in NVS by default. If we read the partition after saving a credential I get the following

```
# Read the NVS parititon on ESP32
esptool.py --port /dev/ttyUSB0 read_flash 0x9000 0x5000 nvs.img

# Read the key values pairs for wifi sta
python3 /home/daigo/src/espressif/esp-idf/components/nvs_flash/nvs_partition_tool/nvs_tool.py nvs.img  -d minimal -f text | grep sta
 nvs.net80211:sta.rrm_e = 0
 nvs.net80211:sta.btm_e = 0
 nvs.net80211:sta.mbo_e = 0
 nvs.net80211:sta.bss_retry = 0
 nvs.net80211:sta.trans_d = 0
 nvs.net80211:sta.sae_h2e = 2
 nvs.net80211:sta.scan_method = 0
 nvs.net80211:sta.minauth = 3
 nvs.net80211:sta.ssid[0] = b'\x05\x00\x00\x00REDACTED_SSID\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
 nvs.net80211:sta.pswd[0] = b'REDACTED_PASSWORD\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'
 nvs.net80211:sta.apsw[0] = b'\x05\x01'
 nvs.net80211:sta.chan = 6
 ```

I want to set the ssid and password using the following format

```
key,type,encoding,value
net80211 ,namespace,,
std.ssid,data,string,"SSID"
std.pswd,data,string,"PASSWORD"
```

However, the SSID is stored as a blob type instead of a string type and is prefixed by \x05\x00\x00\x00

Is there a way I can set the wifi credentials in NVS that is compatible with esp_wifi and is human-friendly?

I originally raised the question on the esp32 forum [here](https://www.esp32.com/viewtopic.php?f=13&t=40579) but didn't get resolved.

ConDai, 2024-09-18:
I understand the team is busy. When can I expect an update?

yujonglee, 2024-11-25:
+1 for this.

chegewara, 2024-11-26:
I am trying to help with this issue, but i see some other "problem", which is not actually an issue, just inconsistency.
I have NVS partition generated with espressif tools, it is working great, but decoding it shows that 4 pages are marked as `full` which according to docs means we cant add new keys to that page (why?):
```
Full
Flash storage is in a consistent state and is filled with key-value pairs. Writing new key-value pairs into this page is not possible. It is still possible to mark some key-value pairs as erased.
```
```
Page no. 0, Status: Full
 hardware:productCode = b'CC0407WB\x00'
 ...
more configs here

Page no. 1, Status: Full
// no data here
Page no. 2, Status: Full
// no data here
Page no. 3, Status: Full
// no data here
Page Empty
```

Another question, to help solving OP issue is 
- is it possible to select page we want to write to? because i found that some espressif settings, like wifi and mqtt creds or phy data are stored on page 4

@ConDai  @yujonglee 
Is there any reason you cant prepare NVS partition the way its described in OP
```
key,type,encoding,value
net80211 ,namespace,,
std_ssid,data,string,"SSID"
std_pswd,data,string,"PASSWORD"
```
then read values in the code and to use values to connect AP?

It is how i am doing it in some commercial product right now and it is working.
```
 wifi:ssid = b'esp32\x00'
 wifi:password = b'espressif\x00'
```

closed 2025-05-21 by espressif-bot