espressif/esp-idf #14558: High retransmission rate in interference-free environment

ShengjieCh, 2024-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.2.2

### Espressif SoC revision.

ESP32C3

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

Low packet loss rate and stable WiFi air interface.

### What is the actual behavior?

When ESP32 is in AP mode and connected to a station to transmit data, the data retransmission rate is high and it works on channel 10 without interference.
[image]
[TEST.zip](https://github.com/user-attachments/files/16979605/TEST.zip)

### Steps to reproduce.

When ESP32 is in AP mode and connected to a station to transmit data, the data retransmission rate is high and it works on channel 10 without interference.

ShengjieCh, 2024-09-13:
ssid: DJI-BEACON1-6919
passwd: edCUzVpQFwAx

MaxwellAlan, 2024-10-12:
Hi @ShengjieCh 

Based on the preliminary analysis of the current packet capture, the retransmission packets do not show any obvious characteristics. We suspect that there might be slight wireless interference in the testing environment, possibly from other wireless devices (such as Bluetooth or mobile phones) around the test area. If possible, please provide photos of the testing environment to help us further analyze the issue. If uploading to GitHub is inconvenient, feel free to contact our company's technical support.

AxelLin, 2024-11-05:
@ShengjieCh Any update regarding https://github.com/espressif/esp-idf/issues/14558#issuecomment-2408529483 ?

MaxwellAlan, 2025-03-07:
Thanks for reporting this issue. Feel free to reopen if still have same issues.

closed 2025-03-07 by MaxwellAlan