espressif/esp-idf #14560: MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 28472762 (us). Drop connection...

nitinGangurde, 2024-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.5

### Espressif SoC revision.

esp32

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32

### Power Supply used.

USB

### What is the expected behavior?

D (37202) MB_TCP_SLAVE_PORT: select() timeout, errno = 0. I (37392) MB_TCP_SLAVE_PORT: Socket (#56), accept client connection from address: 192.168.1.103 E (37392) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 28472762 (us). Drop connection...

### What is the actual behavior?

D (37202) MB_TCP_SLAVE_PORT: select() timeout, errno = 0. I (37392) MB_TCP_SLAVE_PORT: Socket (#56), accept client connection from address: 192.168.1.103 E (37392) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 28472762 (us). Drop connection...

### Steps to reproduce.

Downloaded the esp32 esp-idf master and Slave code on different esp32

### Debug Logs.

```plain
D (37202) MB_TCP_SLAVE_PORT: select() timeout, errno = 0. I (37392) MB_TCP_SLAVE_PORT: Socket (#56), accept client connection from address: 192.168.1.103 E (37392) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 28472762 (us). Drop connection...
```

### More Information.

D (37202) MB_TCP_SLAVE_PORT: select() timeout, errno = 0. I (37392) MB_TCP_SLAVE_PORT: Socket (#56), accept client connection from address: 192.168.1.103 E (37392) MB_TCP_SLAVE_PORT: Client 0, Socket(#55) do not answer for 28472762 (us). Drop connection...

alisitsyn, 2024-09-13:
@nitinGangurde,

Could you provide more information on how you use and configure the Modbus stack on your side? Why do you think the behavior is incorrect? According to log, your slave does not get request from master during`CONFIG_FMB_TCP_CONNECTION_TOUT_SEC` seconds and then drops connection. This is expected behavior. Please check if your ESP32 Master is able to connect to the same network and communicate. Please provide the logs of Master and Slave to recognize your problem.
Thanks.

alisitsyn, 2024-12-05:
It is supposed the behavior is related to broken network connection. Planned to be checked later. The issue is closed due to lack of feedback.

closed 2024-12-05 by alisitsyn