espressif/esp-idf #14562: Http client redirect fails if Location header contains relative URL

CarlosDerSeher, 2024-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

esp32 v1.0

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Mini D1 ESP32

### Power Supply used.

USB

### What is the expected behavior?

Http client should be redirected to to relative url (record_list.php) passed by the server with 302 status code.

### What is the actual behavior?

`http_parser_parse_url()` returns with `s_dead` on the character `_` and therefore http_client can't parse the url.

### Steps to reproduce.

I have to say that I changed the `.host` entry below and also `post_data` isn't original, as I can't share the server and login data here.

```
esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
    static char *output_buffer;  // Buffer to store response of http request from event handler
    static int output_len;       // Stores number of bytes read
    switch(evt->event_id) {
        case HTTP_EVENT_ERROR:
            ESP_LOGI(TAG, "HTTP_EVENT_ERROR");
            break;
        case HTTP_EVENT_ON_CONNECTED:
            ESP_LOGI(TAG, "HTTP_EVENT_ON_CONNECTED");
            break;
        case HTTP_EVENT_HEADER_SENT:
            ESP_LOGI(TAG, "HTTP_EVENT_HEADER_SENT");
            break;
        case HTTP_EVENT_ON_HEADER:
            ESP_LOGI(TAG, "HTTP_EVENT_ON_HEADER, key=%s, value=%s", evt->header_key, evt->header_value);

            if (strstr(evt->header_key, "Set-Cookie")) {
            	strcpy(session_cookie, strtok(evt->header_value, ";"));
            	ESP_LOGI(TAG, "found cookie %s", session_cookie);
            }
            else if (strstr(evt->header_key, "Location")) {
            	//strcpy(location, evt->header_value);
[... 260 lines cut ...]
I (5686) wifi:<ba-add>idx:0 (ifx:0, 30:b5:c2:23:0a:e4), tid:0, ssn:0, winSize:64
I (5886) esp-x509-crt-bundle: Certificate validated
I (6886) HTTP_CLIENT: HTTP_EVENT_ON_CONNECTED
I (6896) HTTP_CLIENT: HTTP_EVENT_HEADER_SENT
I (7196) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Server, value=nginx
I (7196) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Date, value=Thu, 12 Sep 2024 12:04:26 GMT
I (7196) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Content-Type, value=text/html; charset=utf-8
I (7206) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Transfer-Encoding, value=chunked
I (7216) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Connection, value=keep-alive
I (7226) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Expires, value=Thu, 19 Nov 1981 08:52:00 GMT
I (7236) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Cache-Control, value=no-store, no-cache, must-revalidate
I (7246) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Pragma, value=no-cache
I (7246) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Set-Cookie, value=PHPSESSID=j9qs7orag5p0dhmc1f2n7mnqfr; path=/
I (7266) HTTP_CLIENT: HTTP_EVENT_ON_HEADER, key=Location, value=record_list.php
I (7286) HTTP_CLIENT: HTTP_EVENT_REDIRECT
E (7286) HTTP_CLIENT: Error parse url record_list.php
I (7296) HTTP_CLIENT: HTTP_EVENT_ON_FINISH
I (7296) HTTP_CLIENT: HTTPS Status = 302, content_length = 18446744073709551615
I (7306) HTTP_CLIENT: HTTP_EVENT_DISCONNECTED
I (7316) HTTP_CLIENT: Finish http example
```

### More Information.

`http_parser_parse_url()` function expects a schema when the location starts with a letter and therefore fails on the symbol `_`

kayzen9, 2024-09-25:
try https://github.com/espressif/esp-idf/issues/13459#issuecomment-2019815901   @CarlosDerSeher

CarlosDerSeher, 2024-09-25:
I've changed toplevel CMakeLists to this

```
# The following five lines of boilerplate have to be in your project's
# CMakeLists in this exact order for cmake to work correctly
cmake_minimum_required(VERSION 3.16)

set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/common_components/protocol_examples_common)

if(${IDF_TARGET} STREQUAL "linux")
    list(APPEND EXTRA_COMPONENT_DIRS "$ENV{IDF_PATH}/tools/mocks/freertos/"
                    "$ENV{IDF_PATH}/examples/protocols/linux_stubs/esp_stubs")
    set(COMPONENTS main)
endif()

include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(esp_http_client_example)

idf_component_get_property(http_parser http_parser COMPONENT_LIB)
target_compile_definitions(${http_parser} PRIVATE "-DHTTP_PARSER_STRICT=0")
```

but the problem still persists.

mahavirj, 2024-09-30:
@CarlosDerSeher 

> http_parser_parse_url() returns with s_dead on the character _ and therefore http_client can't parse the url.

If your problem is indeed related to above parsing failure then `DHTTP_PARSER_STRICT=0` should have helped here. You can try putting some debug prints under `HTTP_PARSER_STRICT` in file `components/http_parser/http_parser.c` to see that the build part is correct here.

If you still run into any issues then please share a minimal self sufficient application that we can use on our side to recreate the issue.

nileshkale123, 2024-10-03:
So, I am closing this issue now. Thanks.

closed 2024-10-03 by nileshkale123