espressif/esp-idf #14566: TinyUSB MSC FAT format fails, flooding log with errors

olicooper, 2024-09-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.0

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Waveshare ESP32 S3 Nano

### Power Supply used.

USB

### What is the expected behavior?

The format/mount procedure should write the sector information in 512 chunks rather than failing or the procedure should fail gracefully.

### What is the actual behavior?

Application fails here: https://github.com/espressif/esp-usb/blob/d32baca40a25966f274d190a71f508d33b799a71/device/esp_tinyusb/tusb_msc_storage.c#L199

With the following logs repeated until the device is rebooted:
```log
...
E (25811) tinyusb_msc_storage: msc_storage_write_sector failed: 0x102
E (25821) tinyusb_msc_storage: Invalid Argument lba(0) offset(0) size(512) sector_size(4096)
...
```

### Steps to reproduce.

Using the MSC demo with SPI flash: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device/tusb_msc

```c++
const tinyusb_msc_spiflash_config_t config_spi = {
    .wl_handle = wl_handle,
    .mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = 0, //CONFIG_WL_SECTOR_SIZE,
        .disk_status_check_enable = false,
        .use_one_fat = false,
    },
};
```

With partition table (16MB flash in QIO mode @ 80Mhz):
```csv
# Name,   Type, SubType,    Offset, Size,   Flags
nvs,      data, nvs,        0x9000, 0x6000,
otadata,  data, ota,        0xF000, 0x2000,
phy_init, data, phy,       0x11000, 0x1000,
app0,      app, ota_0,     0x20000, 2M,
app1,      app, ota_1,            , 2M,
storage,  data, fat,              , 10M,
```

sdkconfig (using default values):
```
CONFIG_FATFS_SECTOR_4096=y
CONFIG_WL_SECTOR_SIZE=4096
CONFIG_TINYUSB_MSC_BUFSIZE=512
```

### Debug Logs.

```plain
I (353) esp_psram: Found 8MB PSRAM device
I (357) esp_psram: Speed: 80MHz
I (361) cpu_start: Multicore app
I (817) esp_psram: SPI SRAM memory test OK
I (826) cpu_start: Pro cpu start user code
I (826) cpu_start: cpu freq: 240000000 Hz
I (826) app_init: Application information:
I (829) app_init: Project name:     TESTMSC
I (834) app_init: App version:      1
I (839) app_init: Compile time:     Sep 12 2024 20:00:00
I (845) app_init: ELF file SHA256:  xxxxxxxxx...
I (850) app_init: ESP-IDF:          5.3.0
I (855) efuse_init: Min chip rev:     v0.0
I (859) efuse_init: Max chip rev:     v0.99 
I (864) efuse_init: Chip rev:         v0.2
I (869) heap_init: Initializing. RAM available for dynamic allocation:
I (876) heap_init: At 3FC97BA0 len 00051B70 (326 KiB): RAM
I (883) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (889) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (895) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (901) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (909) spi_flash: detected chip: winbond
I (913) spi_flash: flash io: qio
I (917) sleep: Configure to isolate all GPIO pins in sleep state
I (924) sleep: Enable automatic switching of GPIO sleep configuration
I (931) main_task: Started on CPU0
I (951) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (951) main_task: Calling app_main()
I (951) main: Initializing wear levelling
I (3041) main: Mount storage...
W (3041) tinyusb_msc_storage: f_mount failed (13)
W (3041) tinyusb_msc_storage: formatting card, allocation unit size=4096
I (3811) main: Storage mounted to application: Yes
I (3811) main:
ls command output:
I (3821) main: USB MSC initialization
I (3821) tusb_desc:
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 8           │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 2           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 1           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4002      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (3991) TinyUSB: TinyUSB Driver installed
I (3991) main: USB MSC initialization DONE
I (3991) main_task: Returned from app_main()
I (4341) main: Storage mounted to application: No
E (4611) tinyusb_msc_storage: Invalid Argument lba(65) offset(0) size(512) sector_size(4096)
E (4611) tinyusb_msc_storage: msc_storage_write_sector failed: 0x102
E (4611) tinyusb_msc_storage: Invalid Argument lba(65) offset(0) size(512) sector_size(4096)
E (4621) tinyusb_msc_storage: msc_storage_write_sector failed: 0x102
E (4631) tinyusb_msc_storage: Invalid Argument lba(65) offset(0) size(512) sector_size(4096)
E (4641) tinyusb_msc_storage: msc_storage_write_sector failed: 0x102
...
```

### More Information.

This bug can be fixed by setting `CONFIG_TINYUSB_MSC_BUFSIZE=4096` (because `size % sector_size == 0`).. but if the user formats the storage device using a larger sector size then it will still fail and I have no control over users formatting the drive.

It seems that the default sdkconfig is not compatible with the MSC demo and it is not clear how to fix the problem (I can't find any documentation about this restriction).

Even if this is expected behaviour, I am not sure why this doesn't fail gracefully? Instead it floods the log with the error seen above.

olicooper, 2024-09-12:
The [wear levelling docs](https://docs.espressif.com/projects/esp-idf/en/v5.3.1/esp32s3/api-reference/storage/wear-levelling.html) state that the default WL sector size is `512` but also says that it is `4096` when used with FATFS?

> The wear levelling component, together with the FAT FS component, uses FAT FS sectors of 4096 bytes, which is a standard size for flash memory

My project made `CONFIG_FATFS_SECTOR_4096=y` and `CONFIG_WL_SECTOR_SIZE_4096=y` by default which conflicts with what I see in the demo: https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/examples/peripherals/usb/device/tusb_msc/sdkconfig.defaults#L10

roma-jam, 2024-09-16:
Hi @olicooper, 

Thanks for the information. 
I can see that the fs was mounted with error (based on the attached log). The USB MSC class provides only a transport layer here to read and write data from the storage. So, if the storage wasn't mounted then any read/write operation will end up with an error. 

Have you tried to run the wear-leveling [example](https://github.com/espressif/esp-idf/tree/master/examples/storage/wear_levelling) itself?
Does it have the same problem on desired settings?

StefanJo3107, 2024-09-28:
Currently experiencing same issue, ran the tinyusb msc example on the esp32 s3 devkit and got flooded with same errors. Any new info on the issue, or way to resolve it?

tore-espressif, 2024-10-31:
Hello everyone, I'm sorry for the delay, we will try to investigate by end of November

vasulya, 2025-02-10:
Has a solution been found yet?

peter-marcisovsky, 2025-03-28:
Hi @olicooper and everyone.

I managed to reproduce the issue:
- using the very same HW as reported (esp32s3 with 16Mb of flash)
- using the same changes in the `partitions.csv` and  `tinyusb_msc_spiflash_config_t` as reported
- with following components versions (esp-idf 5.3.2) :

```
NOTICE: Processing 3 dependencies:
NOTICE: [1/3] espressif/esp_tinyusb (1.7.2)
NOTICE: [2/3] espressif/tinyusb (0.18.0~1)
NOTICE: [3/3] idf (5.3.2)
```

The difference in my case being:

## Storage mounting

You reported that the storage was not successfully mounted in the beginning and had to be formatted
> I (3041) main: Mount storage...
W (3041) tinyusb_msc_storage: f_mount failed (13)
W (3041) tinyusb_msc_storage: formatting card, allocation unit size=4096
I (3811) main: Storage mounted to application: Yes

In my case, the storage was mounted successfully.

## Default configuration

You reported, that you are using default configurations
> sdkconfig (using default values):

>CONFIG_FATFS_SECTOR_4096=y
CONFIG_WL_SECTOR_SIZE=4096
CONFIG_TINYUSB_MSC_BUFSIZE=512

Which is not true. The default configuration in `tusb_msc` example is different, being:

```
CONFIG_WL_SECTOR_SIZE_512=y
```

Please refer to the [sdkconfig.defaults](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/usb/device/tusb_msc/sdkconfig.defaults#L10) from the `tusb_msc_example`

With the default configurations (the real default, from the example) this error does not occur. After changing the sector size from the defatults 512 to 4096, the error starts to occur.

Then I noticed, that you are using `CONFIG_TINYUSB_MSC_BUFSIZE=512`, which is the default value. But please note, that you have increased the  `CONFIG_FATFS_SECTOR` from 512 to 4096, but the `CONFIG_TINYUSB_MSC_BUFSIZE` stayed the same (512). After increasing the `CONFIG_TINYUSB_MSC_BUFSIZE` to 4096 (while having the `WL_SECTOR_SIZE` at 4096), the error does not occur anymore.

FYI, I managed to reproduce the issue even on esp-idf master branch, using `tusb_msc` example without any changes in the code or in the `partitions.csv`, just after changing the sector size from the default value 512, to 4096. Using esp32s3 module with 4MB of flash.

## Conclusion

Could you please:

- Confirm, that the issue does not occur when using the default configuration?
- Confirm, that the issue does not occur, with following settings: `CONFIG_WL_SECTOR_SIZE_4096` `CONFIG_TINYUSB_MSC_BUFSIZE=4096`

Thank you.

closed 2025-04-07 by peter-marcisovsky