espressif/esp-idf #14567: RMT in ULP Risc-V

mike-2020, 2024-09-13:
### Is your feature request related to a problem?

Can I use RMT in ULP RISC-V? This would be useful for controlling other device (e.g. a LED/light) using IR/NEC in super low power mode.

suda-morris, 2024-09-13:
No, because the peripheral is not in a power domain which can be sustained in a "super low power" mode.

sudeep-mohanty, 2024-09-20:
Closing the ticket since there is no support work necessary. In case you have more issues that you'd need help with, please feel free to reopen this ticket or open a new ticket for us. Thank you!

closed 2024-09-20 by sudeep-mohanty