espressif/esp-idf #14569: Unable to add static library to lp core program

Szybet, 2024-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should simply link the symbol and that's it

### What is the actual behavior?

It complains it can't find it, undefined reference

### Steps to reproduce.

I have built by library (with rust esp-lp-hal) the symbol is there, the architecture is also the same, a normal static library, then I added on top of my ulp program `extern void rust_test();`. I tried many things, calling `link_libraries`, also modifying the `__setup_ulp_project` function, by trying to call `add_prebuilt_library`, the regular cmake ways of adding a library too, in various places, the main cmake file, the one in the main dir etc. I see no reason for it to not work, it should regularly grab the symbol and built with it.

Also as to why, this would enable me to use rust memory safety and trait magic to drive the lp core and esp idf would manage the features that aren't implemented there for rust yet

### Build or installation Logs.

```plain
/opt/esp/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/ld: CMakeFiles/ulp_main.dir/workspaces/yatchy-lp-program/lp-core-esp-idf/main/ulp/main.c.obj: in function `main':
/workspaces/yatchy-lp-program/lp-core-esp-idf/main/ulp/main.c:19:(.text.startup.main+0x4): undefined reference to `rust_test'
collect2: error: ld returned 1 exit status
```

### More Information.

It's for the esp32c6 lp core

sudeep-mohanty, 2024-09-13:
Hi @Szybet, We have recently added the support for adding custom libraries to the ULP project. You can see the example [here](https://github.com/espressif/esp-idf/blob/59e18382702b6986be3d3f55e9ac7763c1397cf7/examples/system/ulp/lp_core/build_system/main/ulp/CMakeLists.txt#L28). This change is on the master branch only.

Szybet, 2024-09-13:
Wow, this is awesome, I will try it later. I hope it will work without specifying source files as I don't have them, but this shouldn't be an issue

Szybet, 2024-09-16:
Worked like a charm, huge thanks