espressif/esp-idf #14572: file serving demo wifi keep disconnecting

Tracker647, 2024-09-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

flash the program, the board shoule be a wifi AP, connect the AP than will found serial output show DHCP assign new ip to client, then can access web page by brower enter url 192.168.4.1

### What is the actual behavior?

the board seems become a STA to connect other wifi and keep output ”wifi disconnected“，can not find the wifi id what I set in menuconfig, inaccessible to webpage.

### Steps to reproduce.

1. select file_serving example
2. simple build and flash
...

### Build or installation Logs.

```plain
I (768) wifi:mode : sta (3c:

[2024-09-13 23:41:13.630]# RECV ASCII>
84:27:9f:9e:54)
I (778) wifi:enable tsf
[0;32mI (778) example_connect: Connecting to myssid...[0m
[0;32mI (778) example_connect: Waiting for IP(s)[0m

[2024-09-13 23:41:16.043]# RECV ASCII>
[0;32mI (3188) example_connect: Wi-Fi disconnected, trying to reconnect...[0m

[2024-09-13 23:41:18.438]# RECV ASCII>
[0;32mI (5588) example_connect: Wi-Fi disconnected, trying to reconnect...[0m

[2024-09-13 23:41:20.845]# RECV ASCII>
[0;32mI (7998) example_connect: Wi-Fi disconnected, trying to reconnect...[0m

[2024-09-13 23:41:23.268]# RECV ASCII>
[0;32mI (10408) example_connect: Wi-Fi disconnected, trying to reconnect...[0m

[2024-09-13 23:41:25.676]# RECV ASCII>
[0;32mI (12818) example_connect: Wi-Fi disconnected, trying to reconnect...[0m

[2024-09-13 23:41:28.070]# RECV ASCII>
[0;32mI (15228) example_connect: Wi-Fi disconnected, trying to reconnect...[0m
```

### More Information.

The demo default setting assuming that esp32 http server is a station, but a server should not be AP for devices to connect? Is I selected a wrong demo or incorrect configure?
The board what I using is esp32c3.

nopnop2002, 2024-09-18:
>flash the program, the board shoule be a wifi AP

No.
This sample does not have the WiFi AP-MODE function.
It will always be in WiFi STA-MODE.

Alvin1Zhang, 2024-09-20:
Thanks for reporting, please help check if this helps https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving#wi-fiethernet-connection, feel free to reopen. Thanks.

closed 2024-09-20 by Alvin1Zhang