espressif/esp-idf #14576: ESP32-C6 set itwt wake interval exponent value under 10, print itwt setup failed, err:0x102

Treamn, 2024-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

it is should be set success, and print iTWT params which I set in monitor

### What is the actual behavior?

Acutally，monitor print "itwt: itwt setup failed, err:0x102"

### Steps to reproduce.

1. set itwt itwt wake interval exponent under 10 in SDK menuconfig
2. build and flash
3. monitor print "itwt: itwt setup failed, err:0x102"

### Debug Logs.

```plain
I (735) itwt: sta connect to whyy
I (740) main_task: Returned from app_main()
I (786) wifi:(he)max.RxAMPDULenExponentExtension:2, max.RxAMPDUpre-EOFpaddingBytes:4194303(0x3fffff)
I (787) wifi:(mac)omc_ul_mu_data_disable_rx:0
I (788) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:2
I (794) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:3, nss:1)
I (802) wifi:(phy)rx_mcs_map:0xfffa, tx_mcs_map:0xfffa, stbc_tx:0, bfmer(su:0, mu:0), ldpc:1, max_rate:172
I (811) wifi:(phy)nsts:2, ru_index_bitmap:0x3(242:1, 484:1, 996:0, 2*996:0)
I (818) wifi:(phy)threshold_bits:24, nsts_num:2, ru_num:2, he_cap->len:26, ppe_threshold_len:4(6,11,4)
I (827) wifi:(ppe)RU242, NSTS0, PPE16:0, PPE8:7, nominal_packet_padding:2
I (833) wifi:(opr)len:7, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:0(max_indicator:0), Basic MCS and NSS:0xfffc
I (846) wifi:(opr)len:7, Default PE Duration:4, TXOP RTS Threshold:1023(0 us), ER-SU-Disable:1
I (855) wifi:(opr)len:7, BSS Color:46, disabled:0, Partial BSS Color:0
I (861) wifi:(spr)len:2, ctrl:0x10, PSR Disallowed:0, Non-SRG OBSS PD SR Disallowed:0
I (868) wifi:(spr)len:2, ctrl:0x10, Non-SRG Offset Present:0, SRG Info Present:0
I (879) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (883) wifi:(connect)dot11_authmode:0x6, pairwise_cipher:0x3, group_cipher:0x3
I (890) wifi:state: init -> auth (0xb0)
I (894) sleep_clock: System Power, Clock and Reset sleep retention initialization
I (1125) wifi:state: auth -> assoc (0x0)
I (1230) wifi:(assoc)RESP, Extended Capabilities length:10, operating_mode_notification:0
I (1231) wifi:(assoc)RESP, Extended Capabilities, MBSSID:0, TWT Responder:0, OBSS Narrow Bandwidth RU In OFDMA Tolerance:0
I (1238) wifi:Extended Capabilities length:10, subtype:0x10, 10:16:88:43:c4:80, Multiple BSSID Support
I (1247) wifi:Extended Capabilities length:10, operating_mode_notification:1
I (1255) wifi:(he)max.RxAMPDULenExponentExtension:2, max.RxAMPDUpre-EOFpaddingBytes:4194303(0x3fffff)
I (1263) wifi:(mac)omc_ul_mu_data_disable_rx:0
I (1267) wifi:(phy)ppe_thresholds_present:1, nominal_packet_padding:2
I (1273) wifi:(phy)dcm tx(constellation:0, nss:0), dcm rx(constellation:3, nss:1)
I (1280) wifi:(phy)rx_mcs_map:0xfffa, tx_mcs_map:0xfffa, stbc_tx:0, bfmer(su:0, mu:0), ldpc:1, max_rate:172
I (1290) wifi:(phy)nsts:2, ru_index_bitmap:0x3(242:1, 484:1, 996:0, 2*996:0)
I (1297) wifi:(phy)threshold_bits:24, nsts_num:2, ru_num:2, he_cap->len:26, ppe_threshold_len:4(6,11,4)
I (1306) wifi:(ppe)RU242, NSTS0, PPE16:0, PPE8:7, nominal_packet_padding:2
I (1312) wifi:(opr)len:7, TWT Required:0, VHT Opr Info Present:0, 6GHz Opr Info Present:0, Co-Hosted BSS:0(max_indicator:0), Basic MCS and NSS:0xfffc
I (1325) wifi:(opr)len:7, Default PE Duration:4, TXOP RTS Threshold:1023(0 us), ER-SU-Disable:1
I (1334) wifi:(opr)len:7, BSS Color:46, disabled:0, Partial BSS Color:0
I (1340) wifi:BSS max idle period:292(1000TU), protected keep alive:FALSE
I (1347) wifi:state: assoc -> run (0x10)
I (1350) wifi:(he)ppe_thresholds_present:1, nominal_packet_padding(rx:2, cfg:2)
I (1357) wifi:(trc)phytype:CBW20-SGI, snr:85, maxRate:172, highestRateIdx:0
W (1364) wifi:(trc)band:2G, phymode:3, highestRateIdx:0, lowestRateIdx:11, dataSchedTableSize:14
I (1373) wifi:(trc)band:2G, rate(S-MCS7, rateIdx:0), ampdu(rate:S-MCS7, schedIdx(0, stop:8)), snr:85, ampduState:wait operational
I (1384) wifi:ifidx:0, rssi:-11, nf:-96, phytype(0x3, CBW20-SGI), phymode(0x7, 11ax), max_rate:172, he:1, vht:0, ht:1
I (1394) wifi:(ht)max.RxAMPDULenExponent:3(65535 bytes), MMSS:5(4 us)
I (1422) wifi:(extcap)mbssid:1, enhanced_mbssid_advertise:0, complete_nontxbssid_profiles:0, twt_responder: 1
I (1422) wifi:connected with whyy, aid = 1, channel 1, BW20, bssid = 10:16:88:43:c4:80
I (1428) wifi:cipher(pairwise:0x3, group:0x3), pmf:1, security:WPA3-SAE, phy:11ax, rssi:-11

I (1452) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1502) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
I (1502) wifi:AP's beacon interval = 102400 us, DTIM period = 1
E (1516) wifi:(itwt)expect TWT SP(32) exceeds WD(16384) 10ms
E (1520) itwt: itwt setup failed, err:0x102
I (1519) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:32, batimeout:2000, txa_wnd:32
I (1535) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:32, batimeout:2000, txa_wnd:32
I (1570) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:32, batimeout:2000, txa_wnd:32
```

Treamn, 2024-09-14:
this is the log if set success
[image]
this is the log when set failed
[image]

Treamn, 2024-09-14:
How can set itwt wake interval exponent under 10? like 0

QingzhaoYin, 2024-09-14:
@Treamn Hi～,We can set itwt parameters as follow. In TWT, wake_interval(TWT_SP) = wake_invl_mant * 2^(wake_invl_exp), wake_duration(WD) = min_wake_dura *wake_duration_unit. There is a limition in driver that should be follow, TWT_SP - WD > 10ms. If want set itwt wake interval exponent under 10, need to adjust wake_invl_mant or min_wake_dura.

Treamn, 2024-09-18:
@QingzhaoYin Thank you! I set wake interval exponent under 10 success!

closed 2024-09-18 by Treamn