espressif/esp-idf #14577: ESP_NOW_Peer => onReceive貌似不太正常

lnsane-G, 2024-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

在onReceive的回调函数中的数据貌似有点奇怪？和随机？请看截图
[image]

// ESP_NOW_Broadcast_Peer.h
#ifndef ESP_NOW_BROADCAST_PEER_H
#define ESP_NOW_BROADCAST_PEER_H

#include "ESP32_NOW.h"
#include "lvgl/gui_ctl.h"
#include "logger.h"

// DeviceType
enum DeviceType
{
    READER,     // 读卡器
    GATEWAY,    // 网关
    METER,      // 流量计
    CONFIG_TOOL // 配置工具
};

// 数据结构体
typedef struct struct_espnow_data
{
    int id;                                            // 包编号
    DeviceType type = READER;                          // 设备类型
    char card_id[10] = {0, 0, 0, 0, 0, 0, 0, 0, 0, 0}; // 卡号
    double flowSpeed = 0.0;                            // 流量速度l/min
    bool recvOk = false;                               // 接受成功
    int channel = 0;                                   // 信道
    char mac[18] = "FF:FF:FF:FF:FF:FF";                // Mac

} struct_espnow_data;

// 创建一个新的类继承自ESP_NOW_Peer类
class ESP_NOW_Peer_Class : public ESP_NOW_Peer
{
private:
public:
    // 类的构造函数
    ESP_NOW_Peer_Class(const uint8_t *mac_addr, uint8_t channel, wifi_interface_t iface, const uint8_t *lmk) : ESP_NOW_Peer(mac_addr, channel, iface, lmk) {}

    // 类的析构函数
    ~ESP_NOW_Peer_Class() {}

    const esp_now_recv_info_t *info;

    // 函数用于注册主机
    bool add_peer()
    {
        if (!add())
        {
            Serial.printf("注册广播主机失败\n");
            return false;
        }
        return true;
    }

    // 向对等点设备发送消息
    bool send_message(const uint8_t *data, size_t len)
    {
        if (!send(data, len))
        {
            Serial.printf("单播消息失败\n");
            return false;
        }
        return true;
    }

    // 函数用于打印从主机接收到的消息
    void onReceive(const uint8_t *data, size_t len, bool broadcast)
    {
        for (size_t i = 0; i < 64; i++)
        {
            Serial.printf("%d ", i);
        }
        Serial.println();
    }

    // 显示发送状态
    void onSent(bool success)
    {
        if (success)
        {
            Logger::log("Send Success\n");
            Gui_Add_List_Text(guider_ui.blueCounter_list_Info, "Send Success", 26, COLOR_GREEN);
        }
        else
        {
            Logger::log("Send Failed\n");
            Gui_Add_List_Text(guider_ui.blueCounter_list_Info, "Send Failed", 26, COLOR_RED);
        }
    }
};
#endif // ESP_NOW_BROADCAST_PEER_H

lnsane-G, 2024-09-14:
[image]

zhangyanjiaoesp, 2024-09-18:
> Serial.printf("%d ", i);

Should this be `Serial.printf("%d ", data[i]);` ?

zhangyanjiaoesp, 2024-12-04:
@lnsane-G Do you still have this issue?

closed 2024-12-19 by Alvin1Zhang