espressif/esp-idf #14578: USB Mass Storage Class example Issue with msc_host_vfs_register()

versamodule, 2024-09-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Using IDF v5.3.1 & ESP32-S3

The only change in menuconfig I did was change Panic handler behaviour to print registers and halt. 

When I create a new project and use the USB Mass Storage Class example, I get a issue with

msc_host_vfs_register(msc_device, MNT_PATH, &mount_config, &vfs_handle)

Here is the terminal output:

I (27) boot: ESP-IDF v5.3.1 2nd stage bootloader
I (27) boot: compile time Sep 14 2024 12:33:34
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) qio_mode: Enabling default flash chip QIO
I (39) boot.esp32s3: Boot SPI Speed : 80MHz
I (44) boot.esp32s3: SPI Mode       : QIO
I (49) boot.esp32s3: SPI Flash Size : 8MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c040020 size=0dacch ( 56012) map
I (113) esp_image: segment 1: paddr=0001daf4 vaddr=3fc92a00 size=02524h (  9508) load
I (115) esp_image: segment 2: paddr=00020020 vaddr=42000020 size=317c4h (202692) map
I (152) esp_image: segment 3: paddr=000517ec vaddr=3fc94f24 size=005b0h (  1456) load
I (153) esp_image: segment 4: paddr=00051da4 vaddr=40374000 size=0e9dch ( 59868) load
I (175) boot: Loaded app from partition at offset 0x10000
I (175) boot: Disabling RNG early entropy source...
I (187) cpu_start: Multicore app
I (196) cpu_start: Pro cpu start user code
I (196) cpu_start: cpu freq: 160000000 Hz
I (196) app_init: Application information:
I (199) app_init: Project name:     z
I (203) app_init: App version:      1
I (208) app_init: Compile time:     Sep 14 2024 12:33:25
I (214) app_init: ELF file SHA256:  1600436fb...
I (219) app_init: ESP-IDF:          v5.3.1
I (224) efuse_init: Min chip rev:     v0.0
I (229) efuse_init: Max chip rev:     v0.99 
I (234) efuse_init: Chip rev:         v0.1
I (238) heap_init: Initializing. RAM available for dynamic allocation:
I (246) heap_init: At 3FC95DF0 len 00053920 (334 KiB): RAM
I (252) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (258) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (264) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (271) spi_flash: detected chip: gd
I (274) spi_flash: flash io: qio
I (279) sleep: Configure to isolate all GPIO pins in sleep state
I (285) sleep: Enable automatic switching of GPIO sleep configuration
I (293) main_task: Started on CPU0
I (303) main_task: Calling app_main()
I (303) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:2 
I (313) example: Waiting for USB flash drive to be connected
I (803) example: MSC device connected
ESP_ERROR_CHECK failed: esp_err_t 0x1701 (ERROR) at 0x42009aee
0x42009aee: app_main at J:/ESP_5.3.1/Projects/z/main/msc_example_main.c:284 (discriminator 1)

file: "./main/msc_example_main.c" line 284
func: app_main
expression: msc_host_vfs_register(msc_device, MNT_PATH, &mount_config, &vfs_handle)

abort() was called at PC 0x4037a06f on core 0
0x4037a06f: _esp_error_check_failed at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/esp_system/esp_err.c:49

Backtrace: 0x40375ab6:0x3fc985f0 0x4037a079:0x3fc98610 0x403808d1:0x3fc98630 0x4037a06f:0x3fc986a0 0x42009aee:0x3fc986d0 0x42030f3b:0x3fc98810 0x4037ab0d:0x3fc98840
0x40375ab6: panic_abort at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/esp_system/panic.c:463
0x4037a079: esp_system_abort at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/esp_system/port/esp_system_chip.c:92
0x403808d1: abort at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/newlib/abort.c:38
0x4037a06f: _esp_error_check_failed at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/esp_system/esp_err.c:49
0x42009aee: app_main at J:/ESP_5.3.1/Projects/z/main/msc_example_main.c:284 (discriminator 1)
0x42030f3b: main_task at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/freertos/app_startup.c:208
0x4037ab0d: vPortTaskWrapper at J:/ESP_5.3.1/esp/v5.3.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: 1600436fb

CPU halted.

versamodule, 2024-09-14:
Found the Issue.  USB drive needs to be formatted as FAT.  Mine was formatted with exFAT. 
Be nice if that was in the README file.

roma-jam, 2024-09-16:
Hi @versamodule,

Thanks for submitting the problem.  
There was couple of related tickets: 
- https://github.com/espressif/esp-idf/issues/12142
- https://github.com/espressif/esp-idf/issues/12662

Of course, it is better to have it described in a README, We will add it there ASAP. 
Sorry for the inconvenience.

closed 2024-11-04 by espressif-bot