espressif/esp-idf #14580: ESP32C6 unable to use nimble - fatal error: esp_nimble_hci.h: No such file or directory

rkonopski, 2024-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Load component on  ESP32C6

### What is the actual behavior?

Error is thrown
```
fatal error: esp_nimble_hci.h: No such file or directory
   10 | #include "esp_nimble_hci.h"
```

### Steps to reproduce.

1. Set target platform as ESP32C6
2. Enable bluetooth
3. Enable nimble
4. Build

rahult-github, 2024-09-16:
Hi @rkonopski , 

Can you please share more information, since all IDF examples do compile cleanly for ESP32C6. 

1. Please share your sdkconfig 
2. Is this your own application ? if yes , have you ensured that correct files are getting included ? 
3. Please ensure you have synced to latest code and have all needed codebase

rahult-github, 2024-09-24:
Hi @rkonopski , did you get a chance to check on this further ?

maksms, 2024-10-13:
I confirm.
OK: esp32s , esp32c3
**NO OK:** esp32c2 , esp32c6

rahult-github, 2024-10-14:
Hi @maksms , 

This is expected. This is because esp-nimble-hci.h  includes defination which are needed for VHCI layer between NimBLE host and ESP bluetooth controller.  esp32 / c3/ s3 use VHCI interface, rest chips don't. 

What functionality do you want to do by including esp_nimble_hci.h file ?

maksms, 2024-10-30:
@rahult-github 
thank you, I really don't need this file for my task

rahult-github, 2024-11-03:
Closing issue per above comment . Please feel free to reopen in case issue is still observed.

closed 2024-11-03 by rahult-github