espressif/esp-idf #14581: Unable to connect ESP32 with iPhone

malto101, 2024-09-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-3023-g3c99557eee

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.1)

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-DevKitC-WROOM

### Power Supply used.

USB

### What is the expected behavior?

When I flash the HFP_HF sample,
The expected behaviour is for the ESP32 to connect to my iPhone, Android and MacOS, Linux

### What is the actual behavior?

I was able to successfully connect to Android, MacOS and Linux

but not on iPhone. it shows the following message on the Iphone

> Connection Unsuccessful
> 
> Make sure "ESP_HFP_HF" is turned on and in range

### Steps to reproduce.

1. Follow the readme.md and flash the ESP32 and check the serial port
2. On the iPhone, click on the Bluetooth setting, and try to connect the device named ESP_HFP_HF

### Debug Logs.

```plain
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7112
load:0x40078000,len:15644
ho 0 tail 12 room 4
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3860
entry 0x4008063c
I (31) boot: ESP-IDF v5.4-dev-3023-g3c99557eee 2nd stage bootloader
I (31) boot: compile time Sep 15 2024 18:12:04
I (33) boot: Multicore bootloader
I (37) boot: chip revision: v3.1
I (41) boot.esp32: SPI Speed      : 40MHz
I (46) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (55) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (64) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=203dch (132060) map
I (151) esp_image: segment 1: paddr=00030404 vaddr=3ff80000 size=00018h (    24) load
I (151) esp_image: segment 2: paddr=00030424 vaddr=3ffbdb60 size=04d2ch ( 19756) load
I (164) esp_image: segment 3: paddr=00035158 vaddr=40080000 size=0aec0h ( 44736) load
I (183) esp_image: segment 4: paddr=00040020 vaddr=400d0020 size=79580h (497024) map
I (353) esp_image: segment 5: paddr=000b95a8 vaddr=4008aec0 size=0c710h ( 50960) load
I (384) boot: Loaded app from partition at offset 0x10000
I (384) boot: Disabling RNG early entropy source...
I (397) cpu_start: Multicore app
I (405) cpu_start: Pro cpu start user code
I (405) cpu_start: cpu freq: 160000000 Hz
I (405) app_init: Application information:
I (408) app_init: Project name:     hfp_hf
I (413) app_init: App version:      v5.4-dev-3023-g3c99557eee
I (419) app_init: Compile time:     Sep 15 2024 18:11:49
I (425) app_init: ELF file SHA256:  2d70ca1c0...
I (431) app_init: ESP-IDF:          v5.4-dev-3023-g3c99557eee
I (437) efuse_init: Min chip rev:     v0.0
I (442) efuse_init: Max chip rev:     v3.99 
I (447) efuse_init: Chip rev:         v3.1
I (452) heap_init: Initializing. RAM available for dynamic allocation:
I (459) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (465) heap_init: At 3FFB6BF8 len 00001408 (5 KiB): DRAM
I (471) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (477) heap_init: At 3FFC8310 len 00017CF0 (95 KiB): DRAM
I (483) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (490) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (496) heap_init: At 400975D0 len 00008A30 (34 KiB): IRAM
I (504) spi_flash: detected chip: generic
I (507) spi_flash: flash io: dio
W (511) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (525) coexist: coex firmware version: 89400176a
I (530) main_task: Started on CPU0
I (540) main_task: Calling app_main()
I (550) BTDM_INIT: BT controller compile version [bc393dd]
I (550) BTDM_INIT: Bluetooth MAC: 2c:bc:bb:0c:09:e2
I (550) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
I (1200) BT_HF: Own address:[2c:bc:bb:0c:09:e2]
I (1210) BT_HF: Starting device discovery...
I (1230) BT_HF: event: 10
I (1230) BT_HF: event: 10
I (1230) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1240) BT_HF: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
I (1240) gpio: GPIO[25]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1250) gpio: GPIO[26]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1260) gpio: GPIO[35]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1270) gpio: GPIO[19]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1280) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1290) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 

 ==================================================
 |       Steps to test hfp_hf                     |
 |                                                |
 |  1. Print 'help' to gain overview of commands  |
 |  2. Setup a service level connection           |
 |  3. Run hfp_hf to test                         |
 |                                                |
 =================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
I (1390) main_task: Returned from app_main()
hfp_hf> I (14050) BT_HF: ESP_BT_GAP_DISC_STATE_CHANGED_EVT
W (16680) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (16690) BT_HF: event: 16
W (16780) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (16970) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 32000
I (17140) BT_HF: authentication success: Mozark’s iPhone
I (17140) BT_HF: d4 2f ca df 16 e7
I (17170) BT_HF: event: 21
W (17270) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
I (17540) BT_HF: ESP_BT_GAP_CFM_REQ_EVT Please compare the numeric value: 166477
W (37390) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 32000
E (47540) BT_HF: authentication failed, status:9
W (47620) BT_HCI: hcif disc complete: hdl 0x80, rsn 0x22
I (47620) BT_HF: event: 17
W (48530) BT_HCI: hcif conn complete: hdl 0x81, st 0x0
I (48530) BT_HF: event: 16
W (48620) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 8000
W (48820) BT_HCI: hcif link supv_to changed: hdl 0x81, supv_to 32000
W (49880) BT_HCI: hcif disc complete: hdl 0x81, rsn 0x13
I (49880) BT_HF: event: 17
```

BetterJincheng, 2024-09-19:
@malto101 This can help. Please have a try and let me know if it works. Thanks,
[image]

malto101, 2024-09-22:
Hello @BetterJincheng, Sorry for the delay in my response.
Thank to you This seems to be working,
Is there any reason for the encryption is off in default?Just want to know the reason, Ideally, Most HFP follows just-pair process with encryption and no authentication if I am not wrong.

None the less, Thank you

BetterJincheng, 2024-09-23:
@malto101 Usually, HF client devices are devices with very limited input and output capabilities, such as portable headphones. These devices cannot input or confirm the passkey or PIN code during the authentication process. So, most HF client devices use a just-work authentication method.

closed 2024-09-23 by malto101